OpenHantek/OpenHantek6022 #1: Wrong maximum sampling frequency

Vascom, 2019-04-30:
I can't set maximum sampling frequency to 48Msps for my 6022BL.
After 24Msps it set only 30Msps in single channel mode.

In two channel mode program can set maximum 16Msps instead of 24Msps.

Ho-Ro, 2019-04-30:
That is the correct and intended behaviour due to HW and USB constraints.
USB2 can transfer (theoretically) 480 Mbit/s = 60 MByte/s but due to protocol overhead and other device's activity on USB only half of this bandwidth can be used reliably by Hantek in bulk transfer mode. Therefore I limited the max sampling to 30 MS/s (30MByte/s) for single channel mode (only CH1) and 16 MS/s (-> 32 MByte/s) for dual channel mode (CH1+CH2 or CH2 only). Also this rate already gives some data corruption due to overruns.
If you want to test the result of much too high sampling rate you can change it in: 
https://github.com/OpenHantek/OpenHantek6022/blob/master/openhantek/src/hantekdso/models/modelDSO6022.cpp  line 28 & 32
See also the discussion in this mega thread: http://www.eevblog.com/forum/testgear/hantek-6022be-20mhz-usb-dso/msg661509/#msg661509

closed 2019-04-30 by Ho-Ro

Vascom, 2019-04-30:
Thanks for clarification.

Vascom, 2019-04-30:
May be add this value as parameter in settings?
Because it is very hard to set desired timebase and samplerate simultaneously. It is impossible as I see: at low timebase openhantek always want switch to maximum sampleate.