OpenHantek/OpenHantek6022 #2: GLX warnings

Vascom, 2019-04-30:
When I run progam from command line in linux I see many qglx warnings when switch settings:
```
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (8 8 8 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 8 8 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 8 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
```

Ho-Ro, 2019-04-30:
Please provide more info about your system:
* Video HW, GLX version, QT version
* What does it say if started with `OpenHantek --useGLES`?

Otherwise a shot in the dark:
https://stackoverflow.com/questions/53768408/qt3d-qt-glx-qglx-findconfig-failed-to-finding-matching-fbconfig-warning
You could try to comment out this line: 
https://github.com/OpenHantek/OpenHantek6022/blob/1f36495056c7ab009d835436e11999ed963d8827/openhantek/src/glscope.cpp#L43
-> `// format.setSamples(4); // Antia-Aliasing, Multisampling`

Vascom, 2019-04-30:
With `OpenHantek --useGLES` same thing.

```
Graphics:  Device-1: Intel Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics vendor: Hewlett-Packard driver: i915 
           v: kernel bus ID: 00:02.0 chip ID: 8086:0412 
           Display: x11 server: Fedora Project X.org 1.20.4 driver: modesetting unloaded: fbdev,vesa compositor: kwin_x11 
           resolution: 1920x1200~60Hz 
           OpenGL: renderer: Mesa DRI Intel Haswell Desktop v: 4.5 Mesa 19.0.3 compat-v: 3.0 direct render: Yes
```
libglvnd-glx-1.1.0-4.gitf92208b.fc30.x86_64
Qt 5.12.1

After comment this line I not see this warning.

Ho-Ro, 2019-04-30:
From what I saw by googling quickly this regression starts with update Qt 5.11 -> Qt 5.12. 
I never deep-dived into all this GL code of OpenHantek, so please check if the hack from above gives some screen artifacts because the setting is intended for anti-aliasing purpose. 
Otherwise we can call it a fix.

Vascom, 2019-04-30:
Yes it is antialiasing.
And if it is Qt regression than better don't change this.
See screenshots without and with this feature
[image]
[image]

Ho-Ro, 2019-05-06:
warnings go away with cleanup 770b8f72

closed 2019-05-06 by Ho-Ro