OpenHantek/OpenHantek6022 #4: Need to correct Timebase and Sampleraset seting

Vascom, 2019-04-30:
It is impossible to set desired Timebase and Samplerate simultaneously.

For example I want 100 ns and 12 Msps but it is always switched to maximum samplerate on lower timebases.

This is a very annoying bug and very bad for user experience.

Ho-Ro, 2019-04-30:
Honestly speaking I do not like the idea that the user selects the sample rate - but OpenHantek is the legacy of a handful of developers over the last nine years and therefore some behaviour is difficult to change. 
My use case of a scope is setting voltage/div and time/div to get the interesting view on screen. 
The scope should set sampling rate accordingly (same as for voltage/div set by user -> scope selects appropriate input gain step).
With the limitation that we have a fixed recording length of 20480 bytes, we have to meet the diverging requirements:
* Good detail resolution (that requires the **highest** possible sampling rate).
* Reliable software trigger (we need a long recording time to catch an edge -> that requires the **lowest** possible sampling rate).
* A stable display (that requires as many points on the screen as possible -> high sampling rate but at least you need a recording time of about double length of the screen size -> lower sampling rate).

Vascom, 2019-04-30:
I would not like to rely entirely on automatic settings.
Especially if I can set manually, as on hardware oscilloscopes.

Vascom, 2019-04-30:
Or just remove the Samplerate setting. Or hide it. But leave the opportunity to set the maximum value so that the data does not fail when transferring via USB.

Vascom, 2019-04-30:
Because on 30Msps I still see artifacts like this:
[image]
16Msps has more clear image.

closed 2019-05-01 by Ho-Ro

Ho-Ro, 2019-05-02:
02b6bd4e Now only the second highest rate is selected as maximum default (24 MS/s for CH1 only, 12 MS/s for CH1&CH2) but 30 or 16 can be chosen manually at the cost of reduced stability.