OpenHantek/OpenHantek6022 #7: Normal mode not working as expected

Ho-Ro, 2019-05-05:
And here you can see that Normal mode not working as expected:
[oh_v_1-2019-05-05_09.34.28.zip](https://github.com/OpenHantek/OpenHantek6022/files/3145199/oh_v_1-2019-05-05_09.34.28.zip)

_Originally posted by @Vascom in https://github.com/OpenHantek/OpenHantek6022/issues/5#issuecomment-489395663_

Ho-Ro, 2019-05-05:
You showed your work flow:

1. Attach a stable input signal, e.g. calibration out signal.
2. Set Auto mode.
3. Move trigger level out of the trace.
4. Trigger condition becomes false, trace loses trigger and runs freely.
5. Set Normal mode.
6. Trace keeps running freely.
7. Move trigger level into trace.
8. Trigger condition becomes true, trace snaps into trigger and becomes stable.

The crash at the end of the video is related to #5 and will not be discussed here.

If I understand correctly you consider step 6. as wrong. What would be your expected behaviour of the program? Should the trace freeze or should it vanish?

As I explained in #5 the shown behaviour is (imho) correct:
"Normal" works as follows: 

* Screen stops in "Normal" mode as soon as trigger condition becomes false and a **triggered** trace is already on the screen.
* As soon as trigger condition becomes true the screen is updated.
* Switching from "Auto" to "Normal" with **false** trigger condition doesn't stop the screen (because no **triggered** trace is available).

closed 2019-05-05 by Ho-Ro

Vascom, 2019-05-05:
Please reopen it.

Om step 6 trace should freeze. And run again only by trigger.

Ho-Ro, 2019-05-05:
> In current normal mode we can't catch single pulse or short series of pulses. 

I disagree, e.g. consider an input square wave of 0 V / 2 V (like calibration out) with a 5 V peek appearing every minute. 
* Set Auto
* Set trigger rising, trigger level 3 V
* Set finally trigger mode Normal

You will see the (not triggered - because we do not have an already **triggered trace** before) input in real time but as soon as the 5 V peek occurs the scope will trigger, show the triggered signal and the signal will now freeze (because we have **no trigger condition** but **a triggered trace** on screen) until the next peek will cross the trigger level.

In short:
Auto mode: Update trace unconditionally - if trigger condition is true then trigger the trace, else keep free running.
Normal mode: If not already triggered before update trace unconditionally free running - but if once triggered then freeze and update only on the next true trigger condition.