OpenHantek/OpenHantek6022 #8: Wrong measurement with single channel 1

UweGo, 2019-05-07:
If i set the checkbox only to CH1, the measurement is very noisy. 
There is no problem with CH1 and CH2 set together or only set to CH2. 
I try the latest repository with the 6022BL.
CH1+CH2:
[image]
Only CH1:

[image]

Vascom, 2019-05-08:
What version do you use?

Ho-Ro, 2019-05-08:
> I try the latest repository with the 6022BL.

* Which OS (linux, mac, win), if linux, which distro/version?
* What is shown on top of scope?  `OpenHantek6022 (Build ...) - Device ... (FW ...)`
* Did you use another program with the scope just before? 
* If yes, did you unplug/replug just before starting OpenHantek?
* Do you get some error or warning output if you start the program in a console?
* Input is the calibration output 2V / 1kHz straight into CH1 and CH2 - correct?

My suspicion is that the scope doesn't switch correctly into single channel mode - this requires the custom firmware version 0200 bundled with OpenHantek. But if there is already a FW (also a different version but with the correct USB VID/PID) uploaded to the scope's RAM OpenHantek doesn't upload again. To force the reload just unplug/replug the scope. (For windows it's [more complicated](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md#windows)).

How does single channel mode work?
In single channel mode the scope transfers one byte (CH1) per sample, this allows double sample rate.
In "normal" double channel mode it transfers two byte (CH1,CH2) per sample.
If the scope requests single mode but the scope doesn't switch correctly (due to wrong FW) then it transfer two byte but the scope expects only one byte -> in the second picture the trace is alternating between CH1 and CH2 values -> double number of points -> one period is 2ms instead of 1 ms.

UweGo, 2019-05-08:
> Which OS (linux, mac, win), if linux, which distro/version?

Linux Uwe-Desktop-PC 4.15.0-48-generic #51-Ubuntu SMP Wed Apr 3 08:28:49 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
What is shown on top of scope?
OpenHantek6022 (Build 20190508) - Device DSO-6022BL (FW 0000)

> Did you use another program with the scope just before?

I try it after booting my PC. I doesn't work with on channel (FW 0000) i unplug the scope and plug in again. Now it work with one channel (and display FW 0200)!
The reason was that i used before the original Hantek Software (in VirtualBox). 
After that i have to reconnect the scope. 
Only rebooting the PC doesn't load the new firmware into the 6022BL.

Best regards,
Uwe

closed 2019-05-08 by Ho-Ro