OpenHantek/OpenHantek6022 #10: Add dummy mode

Vascom, 2019-05-25:
Can you add dummy mode? So users without Hantek hardware can see and may be test interface.

As I think it need only dummy input data generator like calibrating output of 6022 hardware.

Ho-Ro, 2019-05-25:
Yeah, I already thought about this enhancement, I will check what's needed to bypass the HW detection when I've got some spare time...

Ho-Ro, 2019-12-10:
I think the proposal is still valuable but after looking at it for one hour it will take a lot of work to connect too many loose ends. When I want to test the program without my scope at hand I either take a cheap saleae LA clone, e.g. from [Lcsoft](https://sigrok.org/wiki/Lcsoft_Mini_Board). It doesn't provide usable values as there is no ADC connected to the EzUSB chip, but at least the GUI works as expected in realtime. And my most portable solution is an old and no more used [DVB-T stick](https://www.linuxtv.org/wiki/index.php/Hama_DVB-T_USB2_Stick) that's also EzUSB based. I define another 6022 model and udev rule with the new "cold" VID/PID and after uploading the FW into RAM this device looks like a 6022 (without useful data).
Maybe a volunteer wants to provide a working dummy mode as a Xmas gift. 👍

marcizhu, 2019-12-10:
What would be required to add this feature?
I haven't looked at the code too much, but maybe adding a fake `DSOModel` object that generates some simple waveforms should be enough.

Another (possibly more difficult option) would be to simulate a USB hardware device from the software itself, so the main program would detect a real device and start using it as such.

Ho-Ro, 2019-12-12:
The fake model is simple, I did this as mentioned above, also creating dummy samples is one line in `hantekdsocontrol.cpp`. But it's more difficult to circumvent the USB device detection and handling.
It starts there: https://github.com/OpenHantek/OpenHantek6022/blob/1b4c0590439db56abe943c7554cafa83c10db8e9/openhantek/src/main.cpp#L134
Solution would be to define a dummy device: `std::unique_ptr<USBDevice> device = dummy_device;`  maybe derived from class `USBDevice` that provides a dummy_model plus some more content, but never triggers any USB communication.
The data transfer happens in `hantekdsocontrol.cpp`, the device is polled by a timer every 25% of expected cycle time:
https://github.com/OpenHantek/OpenHantek6022/blob/1b4c0590439db56abe943c7554cafa83c10db8e9/openhantek/src/hantekdso/hantekdsocontrol.cpp#L104
Please check the [OpenHantek developer documentation](https://github.com/OpenHantek/OpenHantek6022/blob/master/openhantek/readme.md) and feel free to experiment :)

marcizhu, 2019-12-15:
Sure! I'll take a look at it and try to add this feature to the software in the next few days! If I succeed, I'll open a PR :D

Ho-Ro, 2020-05-06:
I implemented a demo mode with edc9a26, call it with `OpenHantek --demoMode` or `OpenHantek -d`.

closed 2020-05-06 by Ho-Ro