OpenHantek/OpenHantek6022 #11: incorrect voltage display

vizkoze, 2019-05-25:
While testing the new openhantek6022 on linux, i noticed that the frequency and voltage readings are not correct in the lower part of the screen.
Frequency is off when the samplerate is too low. e.g; on the calibration squarewave, when you update the timebase, the samplerate automatically reduces to 100kS/s and gives you a frequency of 250Hz. updating the samplerate to 500kS:S shows the correct frequency values.
It would be nice to give the highest possible samplerate when changing the timebase, this would eliminate some interpretation errors.
Also, voltage readings on the last version seems half of what they shoud be. I'm reading 1, 06 V on the ref squarewave, Normally this should read 2Vpp.
on sinewaves, voltages are half of what they are in reality. Older openhantek software gives correct values

Ho-Ro, 2019-05-25:
> the new openhantek6022

Which version?

> when you update the timebase

Which value? The samplerate drops to 100 kS/s when setting the timebase >= 500 µs/div

> It would be nice to give the highest possible samplerate when changing the timebase, this would eliminate some interpretation errors.

Changing the timebase automatically selects the highest possible samplerate (with the exception that the maximum sample rate 30 MS/s in one-channel mode and 16 MS/s in two-channel mode is not selected automatically so that the sampling does not become unstable). But you can always select these values manually if you want more speed. (48 MS/s is not stable due to USB limitations).

> Also, voltage readings on the last version seems half of what they shoud be. I'm reading 1, 06 V on the ref squarewave, Normally this should read 2Vpp.

No, OpenHantek6022 doesn't show peak-to-peak voltage, the readings are V= (DC value, average of the signal), V~ (AC value of the signal) and Veff (RMS value of the signal), connect your multimeter and compare, it will measure the same values. (read more)[https://en.wikipedia.org/wiki/Root_mean_square]

One remark: When opening issues please use the provided "Bug report" or "Feature request" template and give as much info as possible, it helps me to support your issue - thx.

vizkoze, 2019-05-27:
Thank you for the insights on the voltages and samplerates.
I tried version 2.09 and got following error:
Sending control command e6 failed: Pipe error

I have to say i initially got it to work but then for some reason it stopped working.
I tried installing the deb installer, using the tar.gz file, removed both, compiled from source, wont work anymore.
Previous version openhantek-eb33325 still works normally.
Forgive my ignorance, i'm not that savvy when it comes to linux compiling/debugging

Ho-Ro, 2019-05-27:
Pleased disconnect and reconnect the USB plug before using the new version to ensure that also the new firmware is uploaded If old firmware was loaded as part of the old hantek 33325 version. OpenHantek6022 doesn't work with the old FW (as seen by the error message e6). The new firmware version in the top line must bei 0200.

vizkoze, 2019-05-27:
Thank you for your quick feedback. I got it working now.
I removed the config files located under .config/openhantek/ and the graph was visible again,
after unplugging and replugging, i now have fw0200 in the top screen as you mention, no more errors when i launch from commandline.
thank you for your good work and kind responce

Ho-Ro, 2019-05-28:
You're welcome.

closed 2019-05-28 by Ho-Ro