OpenHantek/OpenHantek6022 #12: Libusb error while running make

shreyansgandhi, 2019-06-26:
[ 81%] Building CXX object openhantek/CMakeFiles/OpenHantek.dir/src/usb/uploadFirmware.cpp.o
/home/intellolabs/OpenHantek6022/openhantek/src/usb/uploadFirmware.cpp: In member function ‘bool UploadFirmware::startUpload(USBDevice*)’:
/home/intellolabs/OpenHantek6022/openhantek/src/usb/uploadFirmware.cpp:38:5: **error: ‘libusb_set_auto_detach_kernel_driver’ was not declared in this scope**
     libusb_set_auto_detach_kernel_driver(handle, 1);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/intellolabs/OpenHantek6022/openhantek/src/usb/uploadFirmware.cpp:38:5: note: suggested alternative: ‘libusb_attach_kernel_driver’
     libusb_set_auto_detach_kernel_driver(handle, 1);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     libusb_attach_kernel_driver
[ 82%] Building CXX object openhantek/CMakeFiles/OpenHantek.dir/src/usb/usbdevice.cpp.o
[ 84%] Building CXX object openhantek/CMakeFiles/OpenHantek.dir/src/utils/printutils.cpp.o
openhantek/CMakeFiles/OpenHantek.dir/build.make:1142: recipe for target 'openhantek/CMakeFiles/OpenHantek.dir/src/usb/uploadFirmware.cpp.o' failed
make[2]: *** [openhantek/CMakeFiles/OpenHantek.dir/src/usb/uploadFirmware.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/intellolabs/OpenHantek6022/openhantek/src/usb/usbdevice.cpp: In static member function ‘static UniqueUSBid USBDevice::computeUSBdeviceID(libusb_device*)’:
/home/intellolabs/OpenHantek6022/openhantek/src/usb/usbdevice.cpp:51:5: **error: ‘libusb_get_port_numbers’ was not declared in this scope
     libusb_get_port_numbers(device, (uint8_t *)&v, sizeof(v));**
     ^~~~~~~~~~~~~~~~~~~~~~~
/home/intellolabs/OpenHantek6022/openhantek/src/usb/usbdevice.cpp:51:5: note: suggested alternative: ‘libusb_get_bus_number’
     libusb_get_port_numbers(device, (uint8_t *)&v, sizeof(v));
     ^~~~~~~~~~~~~~~~~~~~~~~
     libusb_get_bus_number
openhantek/CMakeFiles/OpenHantek.dir/build.make:1166: recipe for target 'openhantek/CMakeFiles/OpenHantek.dir/src/usb/usbdevice.cpp.o' failed
make[2]: *** [openhantek/CMakeFiles/OpenHantek.dir/src/usb/usbdevice.cpp.o] Error 1
CMakeFiles/Makefile2:185: recipe for target 'openhantek/CMakeFiles/OpenHantek.dir/all' failed
make[1]: *** [openhantek/CMakeFiles/OpenHantek.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

Vascom, 2019-06-26:
You not installed libusbx-devel?

Ho-Ro, 2019-06-26:
yes, regarding this post: https://github.com/OpenHantek/openhantek/issues/299
You're using ubuntu, correct?
Which version do you use?
Which architecture? (amd64 ?)
Do you have an old or local libusb install?
Please show  the output of `apt-cache policy libusb-1.0-0-dev`
and also `find /usr -name 'libusb.h'`

To all who report issues with the friendly request for attention  - please use the prepared bug report (or feature request) form and give as complete information as possible. This will save us a lot of questions. Thx!

[image]

[image]

[image]

shreyansgandhi, 2019-06-27:
**OS:** Ubuntu 18.04.02,  4.18.0-24-generic #25~18.04.1-Ubuntu SMP Thu Jun 20 11:13:08 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

Output 1:
libusb-1.0-0-dev:
  Installed: 2:1.0.21-2
  Candidate: 2:1.0.21-2
  Version table:
 *** 2:1.0.21-2 500
        500 http://in.archive.ubuntu.com/ubuntu bionic/main amd64 Packages
        100 /var/lib/dpkg/status

Output 2:
/usr/include/libusb-1.0/libusb.h
/usr/local/include/libusb-1.0/libusb.h

Thank you for the support.

Ho-Ro, 2019-06-27:
It's as I suspected you have a (probably older) libusb installation in /usr/local which is used instead of the correct official version. This happens when you build or install software without using the correct installer of your distribution. (As a quick'n'dirty solution for installing own sw I use [checkinstall](http://manpages.ubuntu.com/manpages/bionic/man8/checkinstall.8.html), it's available from system repo).
Please clean up your build environment - I will close this issue because it's not caused by OpenHantek6022.

closed 2019-06-27 by Ho-Ro