OpenHantek/OpenHantek6022 #13: How can I download and run the finished program. I can’t do this because the uploaded file doesn’t run as a ready application on the iMac.

ArakelGaribian, 2019-07-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Ho-Ro, 2019-07-08:
> only opens the Terminal and does something there

Unfortunately, my glass ball is in repair.
Please provide a detailed report -  if possible use the [bug report template](https://github.com/OpenHantek/OpenHantek6022/issues/new?assignees=&labels=&template=bug_report.md&title=) specially designed for this purpose.

Martin

ArakelGaribian, 2019-07-08:
Dear Martin! I can not understand how to run this program on Mac at all. Here I downloaded the openhantek_20190707-1_osx_amd64.tar.gz assembly and what to do with it next? When I run the file from the packs urs> bin> OpenHantek, the terminal opens and the following is displayed in it: Last login: Mon Jul 8 23:14:58 on ttys000
iMac-A: ~ ag $ / Users / ag / Downloads / usr / bin / OpenHantek; exit;
dyld: Library not loaded: /usr/local/opt/qt/lib/QtPrintSupport.framework/Versions/5/QtPrintSupport
  Referenced from: / Users / ag / Downloads / usr / bin / OpenHantek
  Reason: image not found
Abort trap: 6
logout
Saving session ...
... copying shared history ...
... saving history ... truncating history files ...
... completed.

[The process is completed]

Please help me, an oscilloscope is very necessary in my projects, but with the transition to Mac I can not use it.

Ho-Ro, 2019-07-09:
> Library not loaded: /usr/local/opt/qt/lib/QtPrintSupport.framework/Versions/5/QtPrintSupport

Hi, it seems that your installation misses the right libraries or at least doesn't find them on the expected location on your system. Please have a look at the apple [build instructions](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md#apple) where you can find the required libs. 
The pre-built binary you're using is created by the CI tool travis that runs on an their osx server (in the internet), I do not own any apple computer (as well as a no windows system), so I'm not able to test anything for osx. The CI tool (continous integration) has mainly the purpose to check if a new version that I develop on my debian linux system also _builds_ under osx (and windows). 
You can see how the needed libs were loaded during the [last build process](https://travis-ci.org/OpenHantek/OpenHantek6022/jobs/555348530), please have a look at line 85:

    brew update; brew install qt5 fftw binutils

If this doesn't help please consider to build your own version from source, it's not too difficult.
Please check also the osx packages on [macports](https://www.macports.org/ports.php?by=name&substr=openhantek) that were provided by mac user @ra1nb0w.

Martin

closed 2019-07-09 by Ho-Ro