OpenHantek/OpenHantek6022 #14: The program crashes

ArakelGaribian, 2019-07-10:
The program crashes on the MAC OS if the cursor measures the time intervals. I can send a video to the mail.

Ho-Ro, 2019-07-11:
**PLEASE!** provide a detailed report - if possible use the [bug report template](https://github.com/OpenHantek/OpenHantek6022/issues/new?assignees=&labels=&template=bug_report.md&title=) specially designed for this purpose. **THANK YOU!**

I will close future bug reports without a minimum of usable content.

It's only some clicks away

    Issues -> [New issue] -> Bug report [Get started]

I've attached the content for your information.

Bug report
-----
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
**IMPORTANT:** Always unplug and replug the scope if used before with a different application, e.g. sigrok or the original Hantek software.

Steps to reproduce the behavior:
1. Go to '...'
2. Select '....'
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or a video to help explain your problem.

**Computer environment (please complete the following information):**
 - OpenHantek version: [use always the latest commit, e.g. *release v2.03, commit 1a2b3c4, ...*]
 - OS: [e.g. *linux, macOS, windows, ...*]
 - Distribution, version: [e.g. *debian stable amd64, ...*]
 - Video hardware: [e.g. *Intel Xeon ... Integrated Graphics*]
 - OpenGL version: [e.g. *OpenGL 4.5, Mesa 19.0.3*]
 - Qt version: [e.g. *Qt 5.11.x*]
 - C++ compiler version: [e.g. *gcc 8.3.x*]

**Scope device (please complete the following information):**
 - Device [e.g. *Hantek 6022BE, ...*]
 - Program top line [e.g. *OpenHantek6022 (Build ...) - Device ... (FW ...)*]
 - Input signal: [e.g. *calibration out 1 kHz, sine wave 1 Veff @ 50 Hz, ...*]
 - Probe setting: [e.g. *no probe, probe X1, probe X10, ...*]

**Additional context**
Add any other context about the problem here, e.g. the program output if started in a console like xterm.

Ho-Ro, 2019-07-19:
No response is also an answer :(
Nevertheless I investigated the issue and removed some div-by-zero possibilities during the marker handling -> ff01faf
Closed!

closed 2019-07-19 by Ho-Ro