OpenHantek/OpenHantek6022 #15: Voltage cursors and storage

staticmem, 2019-07-28:
I've looked through the manual and played with the software (v2.12).  Does it have Voltage cursors?  If not will this be added?

Also does the SW store samples or only what can be seen?

Ho-Ro, 2019-07-28:
Short answer to part 2: 
The csv file contains 20000 voltage samples and 10000 full spectrum values (0 Hz .. fsample/2).

Long answer to part 1:
Yes, it has voltage cursors. Look at bottom of page 6/10 of the user manual. Select the round target icon and the cursors window will open. Select and enable voltage cursor and draw a rectangle on the voltage trace to measure voltage and time difference, same applies to the spectrum traces. The detailed description is still missing in the user manual (tbd.) - please explore it on your own. 
[image]

closed 2019-07-28 by Ho-Ro

staticmem, 2019-07-29:
Thanks for the detailed an prompt reply.

The cursor after I worked out I had to click on the OFF to enable it to ON works nicely.  That's a nice feature.  I guess what I really should have asked though are there horizontal Voltage Markers.  Drawing out the area with the mouse provides a solution but if having time markers why not also for have for voltage too?

Ho-Ro, 2019-07-29:
>  if having time markers why not also for voltage too?

You have only one time scale for all voltage tracks, but with up to three voltage channels on the screen, you can get three different scales -> therefore the extra cursor section. 
For what kind of measurement would you need voltage cursors? 
In addition, under the trace there are the results of the voltage measurement DC, AC and RMS, which represent the most important characteristic data for audio measurements, because due to the low sampling rates the scope is rather not suitable for high-speed logic.