OpenHantek/OpenHantek6022 #17: Grid dots are difficult to see

staticmem, 2019-07-29:
Depending on the viewing angle the grid dots are very difficult to see.  Can these be made larger using more pixels perhaps with an option in the colour configuration menu?

Ho-Ro, 2019-07-29:
Have you already increased the alpha channel value of the grid in the colour menu? The default value is 0xBF, set it to 0xFF.
The definition of the colours happens in viewsettings.h, e.g. line 32
` QColor(0xff, 0xff, 0xff, 0xff), QColor(0xff, 0xff, 0xff, 0xbf), // border, grid`

staticmem, 2019-07-29:
Thanks. That solved it but I think my default value was lower than what you stated.  Maybe it should be raised?

closed 2019-07-29 by staticmem

Ho-Ro, 2019-07-29:
It was already raised from 0x3F to 0xBF with ff01faf but if you started with an earlier version it has been kept in the automatically stored config file.