OpenHantek/OpenHantek6022 #19: add option to display Vpp

rpoplawski0, 2019-08-25:
First, thank you for you work on this software! It was one of the main reasons I decided to buy 6022BE.

**Is your feature request related to a problem? Please describe.**
Measuring voltage using "cursor" feature requires taking hands off the whatever stuff you're working on and it's actually quite slow process. A quick glance at the bottom of the screen would be all that's needed to find out Vpp.

**Describe the solution you'd like**
Add Vpp measurement (option to customise bottom status bar would be a bonus)

**Additional context**
The lack of Vpp mesurement confused me when I first plugged the oscilloscope. I assumed my Hantek's calibration is way off as the voltage was nowhere near what I was expecting. Only after I read #11 I've realised why the values are different.

Ho-Ro, 2019-08-26:
Hi Radoslaw,
Vpp was the only reading when I cloned the project. I deliberately removed the Vpp display, because it is very sensitive to noise, glitches etc. and also because my measurement tasks are more in the audio range, where the level is more important than the peak-to-peak voltage, which is already recognizable (for the experienced scope user). But you're right, for digital measurements Vpp is helpful. I'm thinking about reintroducing it.
To do this, I have to go back a bit and describe how the measured values are generated. The scope cyclically collects 20000 sample values but displays only a part of them (< 10000). The SW trigger calculates the beginning of the visible area (to get a standing image, because there is no HW trigger). Depending on the sampling rate and the horizontal resolution, between 100 and 10000 sample points are imaged on the screen.
The spectrum as well as the voltages (DC=average value, AC=RMS of the signal **without** DC component, RMS = true rms value = sqrt( DC²+AC²) are calculated over the totality of the 20000 samples as individual peaks are less relevant for the result - this corresponds to the display of a digital multimeter. 
However, Vpp would only have to be calculated from the visible samples to be consistent with the display (so that a glitch outside the visible range does not distort the measurement). But I think it can be done.
Martin

P.S.: Thank you for filling out the feature-request form, you're the first and only to use these documents to clearly structure your request and make it easier for me to develop it - as a bonus, I'll process your request at short notice :)

closed 2019-08-26 by Ho-Ro

rpoplawski0, 2019-08-26:
Fantastic!
and thank you for the explanation.

commit 8ee45837 (2019-08-26): add Vpp display, closes #19