OpenHantek/OpenHantek6022 #24: Add RPI Binary

MstrVLT, 2019-10-09:
A lot of people, including me, would like to see a version for RPI in the releases. 
Thx!

Vascom, 2019-10-09:
Depends on Operating System on your RPI.

Ho-Ro, 2019-10-10:
I guess you use Raspbian, then just follow the [build instructions for debian](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md#linux). I don't use OpenHantek6022 on a Raspi, so I never compiled it. 
Maybe @LameLefty can provide his build too, if he's still on board. 

Otherwise: **Volunteers wanted!**

See also https://github.com/Ho-Ro/OpenHantek6022/issues/2

Ho-Ro, 2019-10-10:
Other idea would be a CI solution - does anyone know an (easy) web based CI setup to build raspbian binaries, similar to e.g. our [travis](https://travis-ci.org/OpenHantek/OpenHantek6022)?

Vascom, 2019-10-10:
It need not only binary build but also packaging to deb. And two build versions: 32- abd 64-bits.
Better way - add it to repo, used by Raspbian.

vizkoze, 2019-10-10:
Hi, i was trying to build it on my old raspberry pi 1B. the first version with only 256MB of memory and ran out of memory while building. I did get to 62% tho :) 
Will try this evening on a 3a+ or 3b with raspbian buster.

Ho-Ro, 2019-10-13:
I've built a quick binary, please try https://github.com/OpenHantek/OpenHantek6022/releases/download/v2.16-rc2/OpenHantek_armhf_v2.16-rc2

vizkoze, 2019-10-14:
Hi Ho-Ro,
i tried building it myself and using your binary but i got the same error:
libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
QEGLPlatformContext: Failed to create context: 3009

tried on raspbian buster with what i read on:
https://raspberrypi.stackexchange.com/questions/61078/qt-applications-dont-work-due-to-libegl
but i didn't have any luck in my short testing period

Ho-Ro, 2019-10-14:
Hi,
did you try to call `Openhantek --useGLES`?
Independent of your issue I've also built some armhf packages for Raspi, available on the [release v2.16-rc2](https://github.com/OpenHantek/OpenHantek6022/releases/edit/v2.16-rc2) page.
HTH Martin

Ho-Ro, 2019-10-15:
`OpenHantek --useGLES` did the trick. 4d503e4 sets OpenGLES as default for arm architecture (RPi). Binary deb, rpm and tgz packages are available under [Releases](https://github.com/OpenHantek/OpenHantek6022/releases).

closed 2019-10-15 by Ho-Ro

vizkoze, 2019-10-16:
I just wanted to report back and thank you for your build. I had to install libopengl0 to make it work.
maybe it was just because I compiled an older version that still was polluting the installation? 
Anyhow, thank you for this wonderful piece of software

Ho-Ro, 2019-10-19:
> I had to install libopengl0 to make it work.

This lib is now also in the list of requirement in *.deb.