OpenHantek/OpenHantek6022 #25: Bundle for Mac OSX with qt, libusb, fftw

mpietruschka, 2019-10-10:
Hey!

Thanks for that wonderful software. It was very easy to use and I am glad you released a Mac OS version!
 
On the first sight, I thought I just have to execute this application and would be able to measure right away. My first try proofed me wrong.

With a little knowledge about software development I figured out quickly, that I need xcode, xcode command line tools, homebrew, qt, libusb and fftw installed. Right after giving away 40 GB of discspace for like 99% unneeded stuff and 1.5 hours later I was able to start openhantek6022.

To be fair, I had no compilation problems happening. Which has been really good!

On the end of the day I wished there had been a simple bundle containing only the things I need. More like 100MB than 40 GB.

QT describes how to do macOS deployment as a bundle... 
https://doc.qt.io/qt-5/macos-deployment.html

Could you please create a simple Bundle for Mac? I offer myself as a volunteer for testing ;)

mpietruschka, 2019-10-10:
Btw. Pulseview is based on qt, too. Maybe you can "compare" some knowledge from the build process over to here :)

Ho-Ro, 2019-10-11:
Hi @mpietruschka mpietruschka,
As I do not own a mac I'm not able to support - the travis build is mainly intended to detect compilation bugs. Wouldn't you like to find a solution? As a mac user I only know @ra1nb0w  who maintains openhantek on macports. Maybe you will have success together.

ra1nb0w, 2019-10-11:
The fastest solution is to use macports. See https://www.macports.org/install.php to install macports and then `sudo port install openhantek` (now at 2.15).
Then you can find the application on `/Applications/Macports`. Note: binary packages are available until 10.14; for 10.15 you still need to build from sources so the process will be quite slow.

Ho-Ro, 2019-10-14:
Hi @ra1nb0w - v2.15 was withdrawn v2.16 is on the way, please read this comment: https://github.com/OpenHantek/OpenHantek6022/pull/22#issuecomment-541694999

ra1nb0w, 2019-10-14:
thank you! I update macports when 2.16 will be available.

Ho-Ro, 2019-10-20:
@mpietruschka 
There is also another mac user working on packaging:
https://github.com/OpenHantek/OpenHantek6022/pull/29/commits/3f6b16c209332357de642a0b2f6e636056c46384
Please be patient.

Ho-Ro, 2019-10-22:
@mpietruschka, @ra1nb0w and @msiegert: Please have a look at the new osx bundles in v2.17-rc4:
https://github.com/OpenHantek/OpenHantek6022/releases/download/v2.17-rc4/openhantek_20191022-2b105c1-1_osx_x86_64.dmg
https://github.com/OpenHantek/OpenHantek6022/releases/download/v2.17-rc4/openhantek_20191022-2b105c1-1_osx_x86_64.tar.gz
Since i have no mac, everything has originated in blind flight with trial and (lots of) error and the help of [travis](https://github.com/OpenHantek/OpenHantek6022/blob/master/.travis.yml).
It can be built, but does it work?

msiegert, 2019-10-22:
Doesn't work correctly :(m but the binary itself seems to work. I dragged the app bundle into /Applications, but nothing happens when you double click afterwards. If I try "open /Applications/OpenHantek.app" in the terminal, I get the following error message:

$ open /Applications/OpenHantek.app
LSOpenURLsWithRole() failed with error -10810 for the file /Applications/OpenHantek.app.

But the binary itself (/Applications/OpenHantek.app/Contents/MacOS/OpenHantek) starts when called from the terminal and works with my DDS120

Not sure what's going on, will investigate. Is there a way I can trigger building a new dmg?

msiegert, 2019-10-23:
Found the problem (after a long long search...)
the created Info.plist file contains an empty CFBundleIdentifier key, which results in the strange error. Please use the attached CMakeLists.txt, which sets the target property MACOSX_BUNDLE_GUI_IDENTIFIER, which is used to fill CFBundleIdentifier.
The new file also sets the app icon.

With that, double clicking the .app file and "open /Applications/OpenHantek.app" work.

[CMakeLists.txt](https://github.com/OpenHantek/OpenHantek6022/files/3759418/CMakeLists.txt)

Ho-Ro, 2019-10-23:
> Is there a way I can trigger building a new dmg?

Please have a look at the complete bundle process; the text below should go into the section `docs/build.md#macosx`, if it works it can be implemented via CMake so that it is done automatically via `make` or as an own target `make osxbundle`, please check:

----
We recommend homebrew to install the required libraries.

    brew update
    brew install libusb fftw qt5 cmake

If you want to build an OSX bundle make sure the option in `openhantek/CMakeLists.txt` is set accordingly:

`option(BUILD_MACOSX_BUNDLE "Build MacOS app bundle" ON)`

After you've installed the requirements run the following commands inside the top directory of this package:

    mkdir build
    cd build
    cmake ../ -DCMAKE_PREFIX_PATH=/usr/local/opt/qt
    make -j2 
    #
    # now the target was created in subdir openhantek 
    # .. either as single binary OpenHantek, then you're done
    # .. or as a bundle if enabled in ../../openhantek/CMakeLists.txt
    # .. but this bundle is still a template as the dynlibs are not yet bundled
    # .. this magic will happen now
    #
    cd openhantek
    # 
    # deploy all necessary Qt dynlibs into the bundle
    /usr/local/opt/qt5/bin/macdeployqt OpenHantek.app -always-overwrite -verbose=2
    #
    # find all other dependencies, and their dependencies, and their... (you got it!)
    # get the tool (TBD: incorporate it into OpenHantek6022 as a submodule)
    curl -o macdeployqtfix.py https://raw.githubusercontent.com/arl/macdeployqtfix/master/macdeployqtfix.py
    python macdeployqtfix.py OpenHantek.app/Contents/MacOS/OpenHantek /usr/local/opt/qt5
    #
    # finally create OpenHantek.dmg from OpenHantek.app
    /usr/local/opt/qt5/bin/macdeployqt OpenHantek.app -dmg -no-plugins -verbose=2
    # 

----

msiegert, 2019-10-23:
I use macports for all the dependencies...
Your scripts almost works. Macports installs everything in /opt/local, so the path for macdeployqt is wrong. I would just remove the absolute path and reply on the PATH variable to find it
And python on my system links to python 3.7 but macdeployqtfix.py requires 2.7. MacOS comes with python 2.7, so I think it would be safe to replace python with /usr/bin/python to make sure 2.7 is used...
After that, the dmg looks good. But I can't really test it since I have all the libs installed...

Ho-Ro, 2019-10-23:
Thx, I've changed the text above accordingly.
I'm struggling with different systems, local installs like yours and the CI setting at travis, where everything is pulled via brew.

Ho-Ro, 2019-10-23:
So - fingers crossed, the [next release candidate](https://github.com/OpenHantek/OpenHantek6022/releases/tag/v2.17-rc5) is on the way (it's the only way to deploy the created artifacts), please check!

msiegert, 2019-10-23:
Looks good! I can drag it into my Application folder and it starts with a double click! Yeah! Nice work!

ra1nb0w, 2019-10-24:
Just a few notes:
- the Application Bundle is not signed therefore the user see _“OpenHantek.app” cannot be opened because the developer cannot be verified._ and need to use `xattr` or right-click->open
- seems that it doesn't start from DMG
- on my macOS it starts only on the second _Open_
- it doesn't support dark mode
- it doesn't support high-dpi; also with
```
<key>NSPrincipalClass</key>
<string>NSApplication</string>
<key>NSHighResolutionCapable</key>
<string>True</string>
```
The first key should be enought
- You should create a link to /Application on DMG
- Insert the license and Copyright into Info.plist

OS Information:
- macOS 10.15 19A602
- Xcode 11.2 11B44

Anyway, good work ;-)

Ho-Ro, 2019-10-25:
@ra1nb0w 
> ... and need to use `xattr` ... 

`xattr -rc OpenHantek.app`??

The [last osx build](https://travis-ci.org/OpenHantek/OpenHantek6022/jobs/602976046) showed this:

    xattr -l OpenHantek.dmg
    com.apple.FinderInfo:
    00000000  64 65 76 69 64 64 73 6B 00 00 00 00 00 00 00 00  |deviddsk........|
    00000010  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
    00000020

ra1nb0w, 2019-10-26:
You are using MACOSX_BUNDLE therefore to modify/add keys you need to create a new local template. At this point you can add the following text
```
<key>NSPrincipalClass</key>
<string>NSApplication</string>
<key>NSHighResolutionCapable</key>
<string>True</string>
```
see [MACOSX_BUNDLE_INFO_PLIST](https://cmake.org/cmake/help/v3.4/prop_tgt/MACOSX_BUNDLE_INFO_PLIST.html)
About dark-mode: it is strange because you are using Qt 5.13 (be carefull that it supports only macOS >=10.12).
I suggest you to include into the DMG a README.txt in which you describe the actual limits, how to run the application (copy in /Applications and use right-click->open for the first time) and maybe how to help to improve the packaging.

mpietruschka, 2019-10-27:
Wow, this started way faster as I used to expect form github :)))

I didn't guess smart enough to install the whole openhantec through homebrew. But I'm glad there are so many people knowing how to build Qt-Apps for mac. I'd only be able to offer java.

I testet this: https://github.com/OpenHantek/OpenHantek6022/releases/tag/v2.17-rc5

After appoving to execte it does nothing. But when the MacOs/OpenHantek in the package runs well!

It seems to draw the SelfContained Qt-Libraries. At least I disable my local ones.

ra1nb0w, 2019-10-28:
> @ra1nb0w I agree this should be made clear. Anyways, what's the reason for this?

the right-click, if you are asking about that, is to open an application without code signature (otherwise you need to use `xattr -d com.apple.quarantine`).
Have you tried to copy the `.app` away from DMG?

Ho-Ro, 2019-10-28:
@mpietruschka I'm ooo this week - I can come back to you next week earliest.

mpietruschka, 2019-10-28:
@ra1nb0w is xattr -d com.apple.quarantine equal to allow the app through system settings -> security -> open? Because I did that before and it did not start before I did the right click which behavior I have never seen before.

ra1nb0w, 2019-10-28:
@mpietruschka Probably yes but I am not sure.

Ho-Ro, 2019-11-06:
Hi, next version available for testing - please comment!

Ho-Ro, 2019-11-07:
Quick check of `OpenHantek.app` -> One thing that didn't work is the use of own Info.plist template (I used the wrong cmake syntax, must be `set_target_properties(...)` instead of `set(...)`) :(
So these entries are still missing:

    <key>NSPrincipalClass</key>
    <string>NSApplication</string>
    <key>NSHighResolutionCapable</key>
    <string>True</string>

Ho-Ro, 2019-11-15:
@mpietruschka, @ra1nb0w and @msiegert: 
Please have a look at the new osx bundles:
https://github.com/OpenHantek/OpenHantek6022/releases/download/3.0.0/openhantek_20191115-037731c-1_osx_x86_64.dmg
https://github.com/OpenHantek/OpenHantek6022/releases/download/3.0.0/openhantek_20191115-037731c-1_osx_x86_64.tar.gz

Thx for testing!

ra1nb0w, 2019-11-16:
After copying the App outside dmg it works fine; retina (HiDPi) ok.
I also updated openhantek on macports.

thank you for your effort!

Ho-Ro, 2019-11-16:
Good things come to those who wait ;)
It has taken some time and I've learned a lot about MacOSX, but I think I will continue to work happily with Linux.
Martin

closed 2019-11-18 by Ho-Ro