OpenHantek/OpenHantek6022 #26: Initially increase timebase program crashes  (v2.15)

fire1, 2019-10-11:
Release-b106 (v2.15)
Program crashes when initially increase timebase above 20ms on Windows 10. Tested several times, only when timebase is increased very slow and patiently program works fine until restart and start again initial increase of timebase. 
On waveform screen appears some peak to peek noise just before crash.

**To Reproduce**
Start the program  on windows 10 and start increasing timebased. On average fast clicking "up" above 2ms program  freez and close it self.

On previous version everything works fine.

closed 2019-10-11 by Ho-Ro

Ho-Ro, 2019-10-11:
Thx for reporting this - I do test the win build only occasionally, mostly for the UI look and feel. I don't have a windows machine, only a virtual win7 on my linux laptop - exactly for this kind of visual check. On my linux I never saw a crash following your bug description, only this kind of noise. 
The reason for the crash was an array overrun: when increasing the timebase quickly while there is still an old sample in the buffer (with faster sample rate) the graphic subsystem calculates the number of samples (timebase * sample_rate * number_of_divs_on_screen) too high and tries to display these too many points -> garbage and crash (at least on windows). 
I've withdrawn v2.15 and issued v2.16-rc1 as replacement.
Martin

fire1, 2019-10-12:
I cannot describe my joy at the existence of OpenHantek6022. I was using official app for several months, tried several times OpenHantek but without any useful work (triggers does not work at all).

But work done in here over the OpenHantek6022 is way more above than my expectation of the device.
 Great job!!!

commit 40d6734c (2019-10-11): fix #26 - array overrun when increasing time base setting
windows binary crashes, linux binary on my machine never showed this issue
also some small trigger improvements