OpenHantek/OpenHantek6022 #28: Error launching on RPI4

cinosh07, 2019-10-16:
I did  this to install:

`sudo dpgk -i openhan.....`

When I try to start the connection widow appear and then program crash with this message:

QEGLPlatformContext: eglMakeCurrent failed: 3009
QOpenGLFunctions created with non-current context
Segmentation fault

Regards

Cinosh07

Ho-Ro, 2019-10-16:
Which version did you use? Please try latest v2.16 and install the missing lib before with
sudo apt install libopengl0

cinosh07, 2019-10-16:
Already tried all of those :-(

Ho-Ro, 2019-10-17:
Strange, this _3009_ error looks similar to `QEGLPlatformContext: Failed to create context: 3009` from #24 that went away when using _OpenGL ES_. I've checked the build on two Raspi 3 systems, one to create the binary and tested on the other, both with a quite new raspian buster. 
But both systems are headless, build via SSH and testing via VNC from my laptop.

Ho-Ro, 2019-10-23:
Some more RPi debs were produced in the meantime and are available in the releases - please give them another try. 
Are there any other RPi4 user with this issue - or better, with a success feedback?
(Feedback from RPi3 or even older gear is also welcome.)

Tarbal, 2019-10-26:
v2.16 runs just fine on a 3B+

Ho-Ro, 2019-11-06:
Ok, no more feedback - looks like there is no more pain.
I close for now, feel free to comment.

MikeDK, 2020-01-29:
Some Feedback from another RPI4 User...:

Tried today a build with git rev 2f09f614f221fa3a6d4beaf5dc41093cbf10fb7f and also prebuilt binary package openhantek_20191213-414988d_armhf.deb with no luck, I get following output:

```
QEGLPlatformContext: eglMakeCurrent failed: 3009
QOpenGLFunctions created with non-current context
Speicherzugriffsfehler
```

--useGLES has no effect, I get the same error both when used or not.

MikeDK, 2020-01-29:
Ok, a little follow-up .... I was curious if it could have to do with the fact that I use the 'OpenGL desktop driver with fake KMS' (dtoverlay=vc4-fkms-v3d enabled in config.txt) ... this is because kodi won't work otherwise on the rpi4.

Indeed - when I switched to the legacy non-GL driver, it works (although I get the libEGL warning: DRI2: failed to authenticate, but it works)

Ho-Ro, 2020-01-29:
> --useGLES has no effect, I get the same error both when used or not.

GLES is automatically selected for ARM architecture. 

    #ifdef __arm__
        // HACK: Raspberry Pi crashes with OpenGL, use always OpenGLES
        useGLES = true;
    #endif

You could try to comment out [this line](https://github.com/OpenHantek/OpenHantek6022/blob/2f09f614f221fa3a6d4beaf5dc41093cbf10fb7f/openhantek/src/main.cpp#L106)

I will reopen this issue but do not follow up as I do not own a RPI4.

MikeDK, 2020-01-29:
ok I commented the line and tried ... crashes too, but there is a slight difference.
If I start with --useGLES, then I get the following output (this time more verbose as I started it on the pi itself this time)...:
```
pi@osc-pi:~ $ OpenHantek --useGLES
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
QEGLPlatformContext: eglMakeCurrent failed: 3009
QOpenGLFunctions created with non-current context
Speicherzugriffsfehler
```

now when I start without gles, then I get following output:
```
pi@osc-pi:~ $ OpenHantek
qt5ct: using qt5ct plugin
QEGLPlatformContext: Failed to create context: 3009
qt5ct: D-Bus global menu: no
QEGLPlatformContext: Failed to create context: 3009
QOpenGLFunctions created with non-current context
Speicherzugriffsfehler
```

So with the fkms driver and GL there is this extra "3009" line .... the first one is shown when the "search for devices" dialog gets visible, the second line comes as soon as it tries to draw the main window.

So, for the moment, as it is working in legacy non-gl mode, I can live with that, as I do not need kodi on that particular pi.

I guess I try again after some time when there is improvement in the (f)kms driver

Ho-Ro, 2020-02-06:
Maybe this change will help (once it reaches raspbian): https://www.khronos.org/conformance/adopters/conformant-products/opengles#submission_882

MikeDK, 2020-02-07:
Yes this looks promising ... I already compiled and tested that on a kodi-pi, next thing will be to do that on the pi with the hantek ... I'll post my results here as soon as I have some spare time ;)

kevinthefixer, 2020-03-01:
Trying to set up a Pi4 myself.  It does not have Kodi installed, it does have the latest Raspian up-to-date.  Installed the .deb from the latest OH-6022 release today.  Here is what happens when I launch from terminal:

kevin@raspberrypi:~/Downloads $ OpenHantek
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
QEGLPlatformContext: eglMakeCurrent failed: 3009
QOpenGLFunctions created with non-current context
Segmentation fault
kevin@raspberrypi:~/Downloads $ 

kevin@raspberrypi:~ $ OpenHantek --useGLES
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
QEGLPlatformContext: eglMakeCurrent failed: 3009
QOpenGLFunctions created with non-current context
Segmentation fault
kevin@raspberrypi:~ $ 

I'm not sure what to do with the Khronos link above, just seems to be some sort of summary.  I'm certainly willing to help troubleshoot the issue.  I also have a Pi3B+ but it has Kodi installed.  FWIW I have a very similar issue with the Espotek Labrador board;  it works fine on the Pi3.  I have an e-mail in to Chris at Espotek.  Let me know if I can help!

Thanks, Kevin  
kevinthefixer@hotmail.com

MikeDK, 2020-03-02:
@kevinthefixer if you do not have kodi installed on that pi4, then you can set the legacy gl driver in raspi-config/advanced/gl driver.... then the error should disappear

Ho-Ro, 2020-03-02:
@kevinthefixer Up to the Raspi 3B+ the program works, the only known issues are with Pi4...
The content of the Khronos link above was changed recently by Khronos, iirc they reported earlier that they will adapt OpenGL also to the Pi4 line.
And regarding the *--useGLES* option, this is set automatically on the ARM version of OpenHantek6022, so your both versions above are the same.

kevinthefixer, 2020-03-03:
Thanks, guys.  Chris (admin@espotek.com) says it's a known bug in the RPi4 kernel.  He provided this:
https://github.com/raspberrypi/linux/issues/3136
So I'm thinking I need to do some searching on the Pi forums, but it's not going to be tonight, I'm beat!  Thanks for getting back to me so quickly.
-Kevin

straycatsgp, 2020-04-02:
Sorry if this is in the wrong thread but I can maybe contribute.
I just installed OpenHantek on my Raspberry 4 B+ 4GB and it works..
I started it up with the 'OpenHantek --useGLES' command in Terminal and everything worked...
But when I now look in terminal I see several messages after the start up:
qt5ct: using qt5ct plugin
[...]
(process:14568): Gtk-WARNING **: 14:57:33.288: Locale not supported by C library.
	Using the fallback 'C' locale.
libpng warning: iCCP: known incorrect sRGB profile
stop CPU
... WROTE: 792 bytes, 10 segments, avg 79
reset CPU
2nd stage: write external memory
stop CPU
2nd stage: write on-chip memory
... WROTE: 5092 bytes, 68 segments, avg 74
reset CPU
libpng warning: iCCP: known incorrect sRGB profile
qt5ct: D-Bus global menu: no
compile shaders
compile shaders
QEGLPlatformContext: eglMakeCurrent failed: 3009
QEGLPlatformContext: eglMakeCurrent failed: 3009
QOpenGLVertexArrayObject::destroy() failed to make VAO's context current
[...]

I don't know if any of these messages indicate a problem, but since the application works I don't dare to change anything.
What do the last 3 lines mean?

Many Thanks, btw for making Openhantek available for RaspBerry Pi. I used it on my Mac before, but now that it works on RP I can make a dedicated Oscilloscope without having to run from my Mac

Ho-Ro, 2020-04-02:
@straycatsgp 
> I just installed OpenHantek on my Raspberry 4 B+ 4GB and it works..

How did you install the programm? The error messages you posted are from [openhantek](https://github.com/OpenHantek/openhantek) (the legacy application) and not from [OpenHantek6022](https://github.com/OpenHantek/OpenHantek6022). 

> What do the last 3 lines mean?
These error messages are thrown by the Qt [EGL](https://www.khronos.org/egl) sybsystem, which is the interface between OpenGL and the PI's native graphic system, the Broadcom drivers. The behavior of the graphics system has obviously changed between RPi3 and RPi4; since I don't use RPi4, I cannot support this system.

straycatsgp, 2020-04-02:
@Ho-Ro 

I installed it following advice by _'eitriv'_ in another thread (Compatibility with Raspberry Pi #103):
[...]
To summarise installation process on raspberry pi:
Install latest raspbian image (debian stretch).
Start raspi-config with: sudo raspi-config
and select "OpenGL with Fake KMS" under Advanced options.

install additional required packages

sudo apt install:
g++ cmake qttools5-dev qttools5-dev-tools libfftw3-dev binutils-dev libusb-1.0-0-dev
libqt5opengl5-dev mesa-common-dev and libgl1-mesa-dev freeglut3 freeglut3-dev
download openhantek-master.zip from:
https://github.com/OpenHantek/openhantek
extract the zip in your pi home directory.

copy the file firmware/60-hantek.rules to /lib/udev/rules.d/

run the following commands inside the extracted directory:

mkdir build 

cd build 

cmake ../ 

make -j2
Optionally install the program:

sudo make install
plug your device.

start from the command line like this: OpenHantek --useGLES.
[...]

Sorry I wasn't aware I was using the legacy app.
In any case, the application works great on the Raspberry Pi 4B -4GB

I will now try and make it work as a dedicated RP with a touchscreen.

straycatsgp, 2020-04-02:
Just a clarification on the installation I did. 'eitriv' states "start raspi-config with: sudo raspi-config and selct "OpenGL with fake KMS" under advanced options, but in the latest Raspbian there was no such choice.
I ignored this and still everything worked

kevinthefixer, 2020-04-02:
"Install latest raspbian image (debian stretch)" -- This is all obsolete!  The latest Raspian is Buster-based (Debian 10), not Stretch.  I used the .deb installer (ARMHF architecture) rather than attempting to compile it myself, I've had poor luck with that.  I've also had poor luck searching the RPi forums, it seems odd but very few people seem to be using o-scopes with Pi4s.  Perhaps I'll try 1) installing the packages @straycatsgp lists, then 2) compiling (or attempting to compile) it myself.  I wonder if adding those dev packages might allow the Labrador software to work also?  

@HoRo, except for this bug, the Pi4 is a great machine, recommend you grab one, you won't be sorry!

kevinthefixer, 2020-04-19:
There has been a development on the RPi forums that may fix this.  It has to do with the Espotek Labrador board, but as I said earlier, it appears to be the same problem.
[https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=267331](url)

Ho-Ro, 2020-04-29:
I did some experiments today on my 3B+ and tried the three possible GL driver settings:

`sudo raspi-config`
[image]
[image]
[image]

only the 1st setting `G1 Legacy        Original non-GL desktop driver` worked for `OpenHantek6022`, the other two gave the same error as above:

    QEGLPlatformContext: eglMakeCurrent failed: 3009
    QOpenGLFunctions created with non-current context

QuantumFluxCapacitor, 2020-04-30:
Hello folks, I just want to chip in.

I got the exact same error as the comment above (Ho-Ro) on my RPi 4B.

By selecting the Legacy non-GL driver, I got OpenHantek6022BE to work.

FYI.

kevinthefixer, 2020-05-01:
On my Pi4 I only had two options, legacy and one OpenGL (forget which).  What gave me problems was that you cannot get to these options through the GUI config tool, only through sudo raspi-config.  I haven't reversed the procedure I cited above (I still need it for the Labrador board), but since I seemed to get closer using the USB-C port in host mode I left it that way.  I'm guessing rather than testing, but with the two fixes IT WORKS!  
[image]
I did have to launch as root the first time, maybe root owns the usb-c hub?  Minor problem anyway.  IMHO we can mark this issue solved.

kevinthefixer, 2020-05-03:
And an update:  feeling guilty about guessing not testing, I went ahead and just plugged the 6022bl into a usb3 port on the Pi--and it worked just fine, I didn't even need to launch via sudo.  Once again I think we can mark this issue solved, and thank you all!

Ho-Ro, 2020-05-03:
Thanks, your findings will be included in future documentation.

closed 2020-05-03 by Ho-Ro