OpenHantek/OpenHantek6022 #30: High CPU utilization

iss000, 2019-10-19:
Recent versions run with very high CPU utilization.

To reproduce start OpenHantek and use under Linux
`top  -p $(pidof OpenHantek)`

CPU goes above 200%
[image]

I've tracked back the issue and found that it appears after v.2.13 - exactly after commit #94bd5bab476087db293237940980008be7afe807 in file hantekdsocontrol.cpp
[https://github.com/OpenHantek/OpenHantek6022/commit/94bd5bab476087db293237940980008be7afe807#diff-2bbc65588ee9d838415eb592ec2614e9](url)

After manually revert code:
`void HantekDsoControl::updateInterval() {
  // Revert "faster display refresh" 
#if 0
    // after commit 94bd5bab476087db293237940980008be7afe807
  
    // Check the current oscilloscope state everytime 25% of the time
    //  the buffer should be refilled (-> cycleTime in ms)
    cycleTime = (int)( (double)SAMPLESIZE_USED * 250.0 / controlsettings.samplerate.current );
    // Not more often than every 2 ms but at least once every 100 ms
    cycleTime = qBound(2, cycleTime, 100);
#else
    // before commit 94bd5bab476087db293237940980008be7afe807
    
    // Check the current oscilloscope state everytime 25% of the time the buffer
    // should be refilled
    cycleTime = (int)((double)getRecordLength() / controlsettings.samplerate.current * 250);
    // Not more often than every 100 ms though but at least once every second
    cycleTime = qBound(100, cycleTime, 1000);
#endif
    //timestampDebug(QString("cycleTime %1").arg(cycleTime));
}`

... now CPU is back to normal ~25%

[image]

I hope this helps. Congrats and thank you again for your great work!

Ho-Ro, 2019-10-19:
I inserted the change specifically so that the display is refreshed more frequently and above all so that it triggers on rare events. I will set the value a little less aggressively; otherwise you can set the optimal value for yourself.
Since I don't need any other CPU-hungry application when performing measurements, the CPU load is not so important to me; on my old ThinkPad T400 with two cores, the worst-case load was 150%, with the new setting it is reduced to 80%.

closed 2019-10-19 by Ho-Ro

iss000, 2019-10-19:
Maybe this will be more important for RPi users. Thanks!

Ho-Ro, 2019-10-19:
Ok, I check if I can put in some `#ifdef` for RPi.

commit 40c6f943 (2019-10-19): fix #30 - less aggressive sampling