OpenHantek/OpenHantek6022 #32: Duty cicle measurement

thalst, 2019-11-08:
Hi,l. First of all, thanks for this amazing software! It definately runs smoother than any other available. The only thing I miss here is a duty cicle measurement, and other minor measurements such as rise and fall times, etc. Is it possible to implement these? I could not uninstall the original software only because I eventually need those measurements.

Ho-Ro, 2019-11-08:
@thalst Good proposal! 
Please have a look at the latest commit 473221e
Now the top line shows the width of the pulse after trigger and the following pulse as well as the duty cycle: /1.00ms\999µs/ (50%)
[image]

You can measure rise / fall time easily with the drafting compass menu on the left.

closed 2019-11-08 by Ho-Ro

thalst, 2019-11-08:
Awesome!!! Another question: do you have some material regarding the spectrum measurement and how to interpret it? I'm having a hard time finding the frequency for a given spike in SPx

Ho-Ro, 2019-11-08:
That's easy, we have e.g. a 2kHz square wave with frequency peaks at 2, 6, 10, ... kHz. The horizontal dock let you select the frequency base for the display of the spectrum, in our case "Frequencybase 1kHz" (1kHz/div), so a peak at division 2, 6, 10.
[image]
You can set the frequency resolution in the same way as the time domain per division -> same frequency, but with "Frequencybase 10kHz"
[image]
The bottom lines display also the fundamental frequency of the measured traces on the right (2.000 kHz).

thalst, 2019-11-08:
I dont know how to thank you. It's a great work. I'll be throwing more ideas in a few days if you don't mind. 
I run a web hosting company. Feel free to contact me if you need a free lifetime web hosting!