OpenHantek/OpenHantek6022 #34: How to import this software in Qt Creator?

thalst, 2019-11-11:
Hi, 

I want to try to implement some features in this program, and experiment with it. What are the steps to be followed when importing this project to QtCreator(Windows)? Mine wont compile with error:

:-1: error: No rule to make target 'all'.  Stop.

It is a fresh installation of QtCreator. Never used it. I would appreciate the help.

Ho-Ro, 2019-11-12:
I don't use QtCreator, but maybe this helps: https://github.com/OpenHantek/openhantek/issues/203

thalst, 2019-11-12:
How do you compile the code? I was just using QtCreator because I saw it has being used in the original OpenHantek page.

Ho-Ro, 2019-11-12:
For linux it is just the easy way, open the source files in an editor of choice, in my case kate (linux / KDE editor).
And then as written in the [build instructions](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md):

    mkdir build
    cd build
    cmake ../
    make -j2

For windows I don't build anything on my own, it is done by the CI tool appveyor: please have a look at the [build log](https://ci.appveyor.com/project/Ho-Ro/openhantek6022).

    set qtpath64="C:\Qt\5.11\msvc2015_64"
set arch64=x64
  - set dir64=build_x64

  - md %dir64%
  - call "C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\vcvarsall.bat" %arch64%
  - cmake -G "Visual Studio 14 2015 Win64" -DCMAKE_PREFIX_PATH=%qtpath64% -DCMAKE_BUILD_TYPE=%configuration% -B%dir64% -H.

Ho-Ro, 2019-11-14:
QtCreator: Do not *import* the project, just *open* the project, i.e. the `CmakeLists.txt` in the project's top directory. QtCreator will parse this file and create a project. Select the *build* directory accordingly and you're done. 

Quick googling found this page: [How to import CMake project in Qt Creator](https://codeyarns.com/2016/01/26/how-to-import-cmake-project-in-qt-creator/).

A quick test on my debian system looked similar but I didn't have time to follow this topic as I don't want to use another IDE/editor - I'm just happy with console / mc / mcedit for small single file edits, kate for larger edits on multiple files and kdevelop as an IDE with project integration; the last two [share the same editor component](http://kfunk.org/2017/01/10/using-kate-for-hacking-on-c-why-dont-you-try-kdevelop/) with identical look'n'feel. 
Nevertheless the import into kdevelop ran smoothly and reflects all classes and declaration-definition links etc. Also makefile creation (automatically via `cmake ../` and build (after defining `make -j2` as build command) work. *git* and  *github* integration was not yet tested.

closed 2019-11-18 by Ho-Ro