OpenHantek/OpenHantek6022 #35: Looping on detecting the 6022BE.

VVelox, 2019-11-15:
**Describe the bug**
Loops on detecting scope. Closing the program and restarting or replugging the scope has no affect.

It says uploading FW, but never gets to that, even if left for a long while.

In the output below the scope is at USB 1.8.* 

The permission denied is expected for those other devices, but even when I did give my use RW permissions to it all to see what would happen, still looped, just opening and closing other devices as well.

```
poll({ 3/POLLIN },1,995)                         = 0 (0x0)
openat(AT_FDCWD,"/dev/usbctl",O_RDONLY,00)       = 15 (0xf)
ioctl(15,USB_READ_DIR,0x7fffffffb8f8)            = 0 (0x0)
openat(AT_FDCWD,"/dev/ugen0.1",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen2.1",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.1",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.2",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen0.2",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.3",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.4",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.5",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.6",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.7",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.8",O_RDWR,00)        = 16 (0x10)
ioctl(16,USB_GET_PLUGTIME,0x7fffffffb8dc)        = 0 (0x0)
ioctl(16,USB_GET_DEVICE_DESC,0x7fffffffb8e0)     = 0 (0x0)
ioctl(16,USB_GET_DEVICEINFO,0x7fffffffbaa0)      = 0 (0x0)
ioctl(16,USB_GET_DEV_PORT_PATH,0x7fffffffba78)   = 0 (0x0)
close(16)                                        = 0 (0x0)
ioctl(15,USB_READ_DIR,0x7fffffffb8f8)            = 0 (0x0)
close(15)                                        = 0 (0x0)
poll({ 3/POLLIN },1,992)                         = 0 (0x0)
openat(AT_FDCWD,"/dev/usbctl",O_RDONLY,00)       = 15 (0xf)
ioctl(15,USB_READ_DIR,0x7fffffffb8f8)            = 0 (0x0)
openat(AT_FDCWD,"/dev/ugen0.1",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen2.1",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.1",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.2",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen0.2",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.3",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.4",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.5",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.6",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.7",O_RDWR,00)        ERR#13 'Permission denied'
openat(AT_FDCWD,"/dev/ugen1.8",O_RDWR,00)        = 16 (0x10)
ioctl(16,USB_GET_PLUGTIME,0x7fffffffb8dc)        = 0 (0x0)
ioctl(16,USB_GET_DEVICE_DESC,0x7fffffffb8e0)     = 0 (0x0)
ioctl(16,USB_GET_DEVICEINFO,0x7fffffffbaa0)      = 0 (0x0)
ioctl(16,USB_GET_DEV_PORT_PATH,0x7fffffffba78)   = 0 (0x0)
close(16)                                        = 0 (0x0)
ioctl(15,USB_READ_DIR,0x7fffffffb8f8)            = 0 (0x0)
close(15)                                        = 0 (0x0)
poll({ 5/POLLIN },1,-1)                          = 1 (0x1)
```

**To Reproduce**
**IMPORTANT:** Always unplug and replug the scope if used before with a different application, e.g. sigrok or the original Hantek software.

Steps to reproduce the behavior:
1. Go to '...'
2. Select '....'
3. See error

**Expected behavior**
It uploads the FW to the device and starts.

**Computer environment (please complete the following information):**
 - OS: FreeBSD 12-STABLE, r354217
 - ports tree: r517355
 - OpenHantek version: ccdceeb 
 - Video hardware: Nvidia
 - OpenGL version: (( waiting for the package for mesa-demos to build so I can check glxinfo ))
 - Qt version: 5.13.22
 - C++ compiler version: Clang 8.0.1

**Scope device (please complete the following information):**
 - Device: Hantek 6022BE
 - Program top line: Select device
 - Input signal: none
 - Probe setting: none

**Additional context**
Add any other context about the problem here, e.g. the program output if started in a console like xterm.

Started in xterm: qt.qpa.xcb: QXcbConnection: XCB error: 148 (Unknown), sequence: 209, resource id: 0, major code: 140 (Unknown), minor code: 20

After running into issues getting this scope to play nice in this and OpenHantek, I tried pulseview. That worked, although it is not useful for oscopes(unless you really want to use it as a 2channel logic analyzer).

Ho-Ro, 2019-11-15:
This can be caused by other USB high speed devices on the **same** bus. 
It looks like you have some other devices on `/dev/ugen1.*`.
I can provoke a stalled startup when there's an USB stick connected to the same bus, swapping to a plug on another bus solves this problem. Under Linux I use `lsusb; lsusb -t` to show the devices. 
@tspspi Please have a look at this issue - thx

Ho-Ro, 2019-11-21:
Follow-up on https://github.com/OpenHantek/openhantek/issues/196 to track the progress here in OH6022, file `uploadFirmware.cpp`:

@yurivict wrote: 
"OpenHantek6022 ... behaves the same way"
"So the error triggered in `UploadFirmware::startUpload` is never reported to the user, and never terminates the "waiting for upload" state.
...
With the line `libusb_set_auto_detach_kernel_driver(handle, 1); libusb_claim_interface(handle, 0);` fails.
Removing the line `libusb_set_auto_detach_kernel_driver(handle, 1);` allows it to work."

@VVelox: Please check if Yuri's solution helps in your case, thx.

Ho-Ro TODO: Do a code-walk-through to check if this also affects Linux, Windows, MacOSX.

VVelox, 2019-11-21:
Howdy!

Sorry about the delay. Will look into this over tomorrow the next day. :)

Ho-Ro, 2019-11-21:
Howdy back - good night!

yurivict, 2019-11-21:
> Ho-Ro TODO: Do a code-walk-through to check if this also affects Linux, Windows, MacOSX.

Just simulate such failure programmatically on any OS and make sure that it is reported to the user properly.

Ho-Ro, 2019-11-21:
> programmatically on any OS...

any: only Linux (debian stable) in my case, so for the other OSs it has to be done on paper ;)

This USB stuff is a legacy from the forked old openhantek project and I hardly touched it so far, even though I turned almost every stone in the scope functionality. But it is always "the right thing"® to check the returned error codes if available.

yurivict, 2019-11-21:
Error reporting should work the same through the UI in any OS: any error during the firmware upload should terminate the "Uploading" state and be reported to the user. So if you just add ```status = LIBUSB_ERROR``` after that operation - this should be reported to the user on Linux too.

Ho-Ro, 2019-11-22:
This was closed automatically by 8e3701f, but this was only error display. 
@yurivict & @VVelox Please check if 38d28a5 solves the looping.

VVelox, 2019-11-22:
@Ho-Ro Was able to sit down and begin doing some testing today.

Seems to play fine as long as initially run it as root for some reason. Otherwise it errors on 'libusb_claim_interface' for some reason.

Using the item below for with devd. The first is the pre FW upload and the second is post post FW upload. The vendor ID changes when that happens.

For some reason it will find the device as as non-root post FW upload. I've checked and device ownership change is being down both times. So the devd rules are definitely working.

```
notify 1000 {
        match "system"          "USB";
        match "subsystem"       "INTERFACE";
        match "type"            "ATTACH";
        match "vendor" "0x04b4";
        match "product" "0x6022";
        action "chown 1001: /dev/usb/`echo $cdev | sed s/ugen//`.*";
};

notify 1000 {
        match "system"          "USB";
        match "subsystem"       "INTERFACE";
        match "type"            "ATTACH";
        match "vendor" "0x04b5";
        match "product" "0x6022";
        action "chown 1001: /dev/usb/`echo $cdev | sed s/ugen//`.*";
};
```

VVelox, 2019-11-23:
@Ho-Ro Derp! My bad. Just realized I copied the wrong ID. 38d28a5 works! <3

Thanks!

commit 8e3701fc (2019-11-22): fix #35 (report errors during FW upload), fix stalled FW upload
closed 2019-11-23 by VVelox