OpenHantek/OpenHantek6022 #36: Horizontal Scroll

thalst, 2019-11-16:
When using single trigger, the spike may not be exactly centered. Consequently, if the time/div is expanded, the signal goes out of screen. 

A good solution would be a horizontal scroll, option added. Zoom function could tackle the problem, but in  this process, another useful feature that could be implemented is a kind of data acquisition menu that allows to specify how much data a user wants to be saved.

.

Ho-Ro, 2019-11-19:
@thalst Please check, no scroll, but trace is always aligned with trigger mark on screen.

closed 2019-11-19 by Ho-Ro

Ho-Ro, 2019-11-19:
The pdf/png export doesn't support this behaviour currently, don't use it now!

Ho-Ro, 2020-01-08:
The pdf/png export fixed since version 3.0.1

commit b13e314c (2019-11-19): fix #36; align trace to trigger on "frozen" trace