OpenHantek/OpenHantek6022 #37: Differential probe X200 X500 X1000 support

v8karlo, 2019-11-18:
I have Hantek differential probe which uses X500.
OpenHantek6022 supports only X1 and X10 probe.
It would be very nice if you add X200 , X500, X1000 probe support. 
Or maybe user can input factor for his own probe.

Ho-Ro, 2019-11-20:
Hi @v8karlo, 
A quick'n'dirty solution is to change line 22 of [openhantek/src/docks/VoltageDock.cpp](https://github.com/OpenHantek/OpenHantek6022/blob/master/openhantek/src/docks/VoltageDock.cpp) to 
`#define ATTENUATION 500`
and line 62 from
`b.attnCheckBox = new QCheckBox(tr("x10"));`
to
`b.attnCheckBox = new QCheckBox(tr("x%1").arg( ATTENUATION );`
I will introduce the 2nd change later into the master branch and make the 1st define selectable via cMake call.
I've already thought and discussed the solution with a drop-down list instead of a single checkbox, but there are so many different probe gains that we would get a huge list, while 95% of all probes are just X1 and X10. (I just added 200 and 500 to my former wish list of 1, 2, 4, 5, 10, 20, 50, 100, 1000... :) )
Another reason is that the Hantek6022 (a low-cost device for beginners and hobbyists) comes with X1/X10 probes and the user interface should supports this without creating [further confusion](https://github.com/Ho-Ro/OpenHantek6022/issues/7). If I have the need for an X100 probe or even a differential probe, then I am an experienced user and can easily convert the values in my head.
BTW: [How was it done in former times?](http://w140.com/tekwiki/images/9/94/Tek_p5200_5.png)

v8karlo, 2019-11-20:
Thank you for your answer!
My differential probe is HT8100 Hantek 50/500, and I have current probe hantek 10/100. All of those values are Hantek.
Even Hantek is for beginners, I made more than few quite professional jobs with it.

Problem arise when I use differential probe with current probe which I do a lot for math graph.

I can do the old way and calculate, but when using math graph it is far easier if attenuation is of correct value and faster.

I am not C++ programmer, but C# mostly.  I will try your solution anyway.

May I suggest possible solution?
Instead of fixed checkBox X10, put the checkBox and textBox where user can enter value for his probe per channel. When checkBox is checked, software read value from textBox and use that value in calculations. 
That way every user can define one value for his own probe type per channel.
No need for long drop down list with lots of values.

Thank you in advance.
This software is gem for eyes after struggling with Hantek software. Very good work!

Ho-Ro, 2020-01-21:
closed by #50 & #51

closed 2020-01-21 by Ho-Ro