OpenHantek/OpenHantek6022 #39: Wrong gain settings for SainSmart DDS120

Ho-Ro, 2019-12-02:
This [report](https://www.eevblog.com/forum/testgear/sainsmart-dds120-usb-oscilloscope-(buudai-bm102)/msg2809852/#msg2809852) from eevblog tells us:
>Have problem with DDS120 and OpenHantek6022 (b188)
I try use my DDS120 again, find new software OpenHantek6022 (b188) and try configure all again.
I see that OpenHantek6022 not read calibration data from DDS120 and zero is not in zero and gain wrong.

Proposed solution: Change these lines:
https://github.com/OpenHantek/OpenHantek6022/blob/c365d9bb8c6e4e17872de031e17391fbcb574cd9/openhantek/src/hantekdso/models/modelDDS120.cpp#L34
https://github.com/OpenHantek/OpenHantek6022/blob/c365d9bb8c6e4e17872de031e17391fbcb574cd9/openhantek/src/hantekdso/models/modelDDS120.cpp#L35

to:

    specification.voltageLimit[0] = { 32 , 80 , 160 , 155 , 170 , 165 , 330 , 820  };
    specification.voltageLimit[1] = { 32 , 80 , 160 , 155 , 170 , 165 , 330 , 820  };

Also adapt https://github.com/OpenHantek/OpenHantek6022/blob/c365d9bb8c6e4e17872de031e17391fbcb574cd9/openhantek/src/hantekdso/models/modelDDS120.cpp#L51

    double calibration = settings.value( ranges[ iii ], "0.0" ).toDouble();

closed 2019-12-02 by Ho-Ro

Ho-Ro, 2019-12-02:
@msiegert Please check how the scope behaves without *.conf file in place. 
Target should be to get a reasonable display on 1st usage without need of special user action.

msiegert, 2019-12-02:
Sorry, but I can't test this right now. My DDS120 is in Japan (had to leave it there to monitor CAN bus communication on a prodction tool I am working on). I bought a LOTO OSC482 as a replacement since the DDS120 seems to be end of life/out of stock. But despite the fact that the USB PID/VID are the same as the DDS120, the configuration is somewhat different from the DDS120 and I had to patch the firmware and OpenHantek6022 a bit to get it to work. I am not sure how to merge my changes to OpenHantek6022 since those will break support for the DDS120...

Ho-Ro, 2019-12-03:
Ok, no problem, I've just removed your 1.18 : 1.21 correction if there is no conf file and therefore adjusted the specification values accordingly, so it should give a +- 5% accuracy for 1st time users - only offset has to be adopted individually. 
Pity that the LOTO has same PID/VID and behaving differently, any chance to check other USB id values?

msiegert, 2019-12-03:
I'll see if I can find a differentiation parameter between the OSC482 and the DDS120 as soon as I get the DDS120 back, but this will be a while. There is some hope, the OSC482 is detected as:

OSC482:
  Product ID:	0x8102
  Vendor ID:	0x8102
  Version:	0.00
  Speed:	Up to 480 Mb/sec
  Manufacturer:	OS19040
  Location ID:	0x14200000 / 25
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

by my Macbook. So we should be able to use the name and/or the manufacturer to distinguish between the DDS120 and the OSC482

alimoche, 2020-03-31:
same issue here with DDS120 (buudai) after updating hantek rules and deleting *.conf file, not working properly, please see attached screenshot, the trigger level is on the top of the screen and in the terminal window says "Couldn't get channel level data from oscilloscope"
[image]

commit c847d81b (2019-12-02): fix #39, better default gain/offset for DDS120