OpenHantek/OpenHantek6022 #42: Software does not utilise native icon theme

Ragnar770, 2019-12-09:
There's a good chance this is a user error but the program doesn't utilise my icon theme which makes buttons harder to see with a dark theme. I've included a screenshot below.

Version: Latest commit (a7c9eb6395986101818436aa09c318b331d5f8d7), built with yay from the AUR
OS: Arch Linux (Kernel 5.4.2)
DE: KDE Plasma 5.17.4 / Qt 5.13.2
Icon theme: La Capitaine

[https://i.imgur.com/p1nAUCi.png](screenshot)

bmentink, 2019-12-10:
HI,
I am having the same issue with icons .. see attached:
[image]

This running Archlinux in a chroot environment ..

Ho-Ro, 2019-12-10:
@Ragnar770 This looks exactly like I expected - what other look would you like to see? Are you talking about the top left icons? 
[image] 
This is what I see
[image]
The left/top run/pause icon comes from the system, the other three right/down (digitalphosphor = sine waves; zoom = search-plus; measure = drafting compass) are hardcoded into the binary, defined in the [source code resources](https://github.com/OpenHantek/OpenHantek6022/tree/master/openhantek/res/images)
I hacked a quick blue digitalphosphor.svg, please try if this approach gives better visibility, then I can provide also the other icons in blue:

    <svg
       xmlns:dc="http://purl.org/dc/elements/1.1/"
       xmlns:cc="http://creativecommons.org/ns#"
       xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#"
       xmlns:svg="http://www.w3.org/2000/svg"
       xmlns="http://www.w3.org/2000/svg"
       xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
       xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
       width="64"
       height="64"
       inkscape:export-xdpi="90"
       inkscape:export-ydpi="90">
      <g transform="translate(1,12.5)" >
        <path
           style="opacity:1.0;fill:none;stroke:#8080ff;stroke-width:4.0;stroke-linecap:round;stroke-opacity:1"
           d="m 1,20 c 13,-10 17,-10 30,0 13,10 17,10 30,0"
    	/>
        <path
           style="opacity:1.0;fill:none;stroke:#4040ff;stroke-width:4.0;stroke-linecap:round;stroke-opacity:1"
           d="m 1,20 c 13,-22 17,-22 30,0 13,22 17,22 30,0"
        />
        <path
           style="opacity:1.0;fill:none;stroke:#0000ff;stroke-width:4.0;stroke-linecap:round;stroke-opacity:1"
           d="m 1,20 c 13,-39 17,-39 30,0 13,39 17,39 30,0"
        />
      </g>
    </svg>

@bmentink No, the icons are linked as a resource file into the binary, as well as the DE, FR, IT, ... translations. 
https://github.com/OpenHantek/OpenHantek6022/blob/b0e709435743b331b127b8b42f45efd4a0779df9/openhantek/src/mainwindow.cpp#L44
You need only one binary (+ the needed system libraries + an udev rule unless you are root) to runthe program.

Do you get some info or error message when starting in a terminal?

bmentink, 2019-12-10:
@Ho-Ro 
Thanks. The only message I get at the console is:
`XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'`

closed 2019-12-10 by Ho-Ro

Ho-Ro, 2019-12-10:
@bmentink: This issue was closed automatically by a commit for the original issue of @Ragnar770, but I'll keep an eye on your problem. Do you have access to a debian system, maybe virtual or from USB stic etc. where you can compare the Qt installation against your more minimal systems?

bmentink, 2019-12-10:
@Ho-Ro 
No, I do not have any other systems, they are all Alpine based ..
Can you give me a clue as to what Qt component I may be missing ..

Ragnar770, 2019-12-11:
I hadn't looked at the source code so I had no idea they were hard coded, that makes a lot of sense actually. The blue icons are a bit easier to see but I'll play around with the vector files and see if I can get them to match my Qt theme, thanks!

bmentink, 2019-12-11:
@Ho-Ro 
I have fixed it my end. I needed to install qt5-qtsvg, suggest this gets added as a requirement check?

Ho-Ro, 2019-12-11:
The blue theme was just a test, but it doesn't please me too.

Ho-Ro, 2019-12-11:
Here are the icons for the dark theme, automatically switching depending on the lightness of the window border (1b4c059)

    QString iconPath = QString( ":/images/" );
    if ( QPalette().color( QPalette::Window ).lightness() < 128 ) // automatic light/dark icon switch
        iconPath += "darktheme/"; // select top window icons accordingly

[image]

Ho-Ro, 2019-12-11:
@bmentink qt5-qtsvg was automatically loaded on my debian system with the other Qt devel stuff, but it is a good hint, I will add it to the build description.

commit 77037a70 (2019-12-10): fix #42, at least the part of @Ragnar770