OpenHantek/OpenHantek6022 #44: Jitter on high frequencies

bmentink, 2019-12-10:
I notice quite a bit of jitter on the edges when the frequency get's higher than about 70khz, is this a hardware limitation or this software ..?

I have tried this on both external signal generator and the internal Calibration output ..

To duplicate just set Cal Out to 100khz ..

Ho-Ro, 2019-12-12:
I checked it with my signal generator, 460 kHz, CH1 only, 30 MS/s, 500 ns/div:
[image]
There is a jitter of ~50 ns that fits with the uncertainty due to sampling every 33 ns. 

If you look at the *calibration out* with a "real" scope you can detect the massive jitter at high frequencies. This signal is created by the firmware running on the scope, interrupt driven by counter 2 running at 4 MHz, for 100 kHz the interrupt is called every 5 µs and toggles an output pin while the 8051 core runs at 48 MHz. Therefor the jitter is expected, but the frequency on long term is quartz stable. I changed calibration out from fixed 1 kHz to the 50 Hz ... 100 kHz range to have a "poor man's function generator" at hand to play around, but never as a precision measurement device.

30 MS/s is the maximal sampe rate that the high speed USB can support - raw USB data rate is 480 Mbit/s -> 60 MByte/s - BUT you have protcol overhead and (very important) the USB bulk transfer shares the bus with other devices and has to claim and release the bus. So from 60 MByte/s only half of it is useful as netto data rate. My program selects the sample rate automatically from chosen time/div setting, but never uses the max. value (30 MS/s for CH1 only, 15MS/s as soon as CH2 is selected, even CH2 mode transfers both channels), but instead 24 or 12 MS/s are selected - but you can select 30/15 manually. 
48MS/s is physically nonsense and only a marketing gag. See also #1

Long story short: Hantek6022BE drives the USB to its limits.

closed 2019-12-12 by Ho-Ro

bmentink, 2019-12-12:
Cool, thanks for the explanation ..