OpenHantek/OpenHantek6022 #45: AC/DC coupling mode

Habakuk2000, 2019-12-18:
Is ist possible to enable the ac/dc coupling without rebuilding the binaries. ex. by an option?

Ho-Ro, 2019-12-19:
I don't provide this option - technical reason: the const structures in `modelDSO6022.cpp`, where the behavior is defined, are defined for all possible models **before** `main()` starts, so an command line option can't change anything anymore. The appropriate device model is later activated using the USB identifier, which does not provide any indication of AC/DC. The firmware doesn't differentiate either, it just won't switch to AC if it's not defined in the model file.
I keep your suggestion in mind, but follow it with low priority - because it works for me as it is. ;).

kiwichrish, 2020-01-25:
? The 6022 didn't have AC coupling mode?

Habakuk2000, 2020-01-25:
with al little modification ... have a look at:

https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/HANTEK6022_AC_Modification.pdf

kiwichrish, 2020-01-25:
ROFL..  I just use a cap in series with the probe externally when I want AC coupling on mine. :-)

So once the mod is done does the re-compiled version of OpenHantek let you turn the relays on/off?  Definitely worth doing..  that schematic explains why the damn thing clips some signals as well, depending on V/div setting..

Ho-Ro, 2020-01-28:
> So once the mod is done does the re-compiled version of OpenHantek let you turn the mos relays on/off?

Sure, that's the purpose of my modification.

> that schematic explains why the damn thing clips some signals as well, depending on V/div setting.

If you take a closer look at the data sheet of the ADC, you will see another design error - the input voltage of the ADC (Ain=-0.5V..+0.5V, /Ain=0) is far outside the specification (common mode voltage: +0.7V..+1.1V). This may be the reason why the 1st 1000..2000 samples are garbage - I just throw them away during sampling.

[image]
[image]

Some more [HW info](https://github.com/Ho-Ro/Hantek6022API/tree/master/hardware) and [FW info](https://github.com/Ho-Ro/Hantek6022API/tree/master/docs).

jm-o, 2020-02-05:
Why not just leave AC and DC options always available on gui?

Ho-Ro, 2020-02-05:
AC coupling requires a hardware modification that the majority of users do not have. The machine should only offer AC coupling if it is also available. Otherwise there will be many irritations and issue reports.

jm-o, 2020-02-05:
Even when choosing DC?
In the source code, it seems that this macro is only used to prevent AC selection.

FYI: As I only have cmake 3.10 (Ubuntu 18.04), I am unable to compile with HANTEK_AC defined. Function add_compile_definitions function has been added in cmake 3.12.

Habakuk2000, 2020-02-05:
@jim-o:
good idea. leaving the AC/DC coupling mode 'always on' but you have to enable it with a checkbox in the settings menu... that should not lead to irritations.
btw: it took me a few days getting "visual studio", "cmake" and especially "qtcreator" to run. but i am unable to create 'release build' with qtcreator. whatever i do, it always creates 'debug build'. no way. using visual studio, everything works fine.

Ho-Ro, 2020-02-06:
@Habakuk2000: 
No need to worry, the debug overhead is removed latest at (linux) package creation. If you want to run the bare binary you can strip it like I do for testing my linux build:

    cd build
    make -j4 && strip openhantek/OpenHantek && openhantek/OpenHantek

That's it - for Windows VS should do the task.
BTW: what is your system?

@jm-o:

> Even when choosing DC?

What do you mean??? 
If you select DC, it will measure with DC coupling, but if you would select AC on an unmodified Hantek, the SW will send the "switch-to-AC" command and the Hantek will set the appropriate port bits (without any effect) and will continue to use DC coupling - **but** the scope screen will tell you "AC coupling". 
Many Hantek6022 users are newbies or *makers* and don't read documentation and will be confused. See the lengthy discussion about probe attenuation (fixed, variable or even at all), I don't need it because I'm used to do the range adaption in my brain since decades :) 

Regarding your Ubuntu cmake version, you can change the entry in `CMakeList.txt` to:

    if ( DEFINED HANTEK_AC )
        if(${CMAKE_VERSION} VERSION_LESS "3.12.0") 
            add_definitions( -DHANTEK_AC )
        else()
            add_compile_definitions( HANTEK_AC )
        endif()
    endif()

(currently untested as I'm away from my development system)

Habakuk2000, 2020-02-06:
@Ho-Ro:
My System is Windows 10 Pro

jm-o, 2020-02-06:
> What do you mean???
I didn't understand what you meant when you said "Otherwise there will be many irritations and issue reports.".
It wasn't software issues but users issues :-)

An AC checkbox in Settings/Scope, disabled by default, seems to be a good enough solution to keep the user informed.

commit df55001d (2020-06-23): close #45; ctrl scroll wheel zooms, scroll wheel shifts zoom range
closed 2020-06-23 by Ho-Ro