OpenHantek/OpenHantek6022 #46: Cursor Measurements yellow/ blue box fails to draw.

LINUX3932, 2019-12-20:
The cursor measurements yellow or blue box fails to draw.

1. Click Compass tool
2. Select a trace
3. Click and drag to draw box.

No box shows. However, the measurement itself is displayed:
[image]

DSO-6022BE FW0204
v 3.0.1 - downloaded binary
Linux Mint 19.2 Cinnamon
Intel Corporation Core Processor Integrated Graphics Controller
OpenGL version string: 2.1 Mesa 19.0.8

terminal output:
$OpenHantek
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no

Ho-Ro, 2019-12-21:
Try setting the trace to ON (click on the OFF).

LINUX3932, 2019-12-21:
Doh! Yes it was there right in front of me. Must have been more tired than usual yesterday. Thanks again Ho-Ro!

closed 2019-12-21 by LINUX3932