OpenHantek/OpenHantek6022 #47: Please overhaul front-end of PDF documentation - FRUSTRATED

www-MKRD-info, 2020-01-01:
Please overhaul the front-end of your documentation relating to satisfying installation dependancies, the installation process, and regarding launching the software.
This is what I got after three hours of effort, and I can't even find the launch shortcut for the software after the three hours, so I will TRY to not swear regarding my dissatisfaction:

1. Launching .deb package binary installer fails with a dependency message. It took me a while to realize that you have pre-installation dependency satisfaction instructions. These instructions must belong to first page of your manual. These instructions must also be prominently all over your documentation, project front page, PDF, and within the .deb launcher.

2. You have a bizarre statement on first page of manual, something about forking, yet don't describe the pre-installation requirements. Replace the bizarre statement with useful information!!!

3. In the PDF manual, please explain to user the installation process with a .deb. Not everyone is a developer here, and not everyone has spent a day getting all the pre-requisites!!!

4. After running your suggested command for dependencies, .deb binary is allowed to install but after installation I can't find in any place any way of seeing what and where was installed, and how to launch it. Again, in your manual, please add wording on how I am supposed to launch your software, what its launch names/executables/location/icons/menu entries look like, etc!!!

Thanks!

Ho-Ro, 2020-01-02:
I am frustrated by error messages where someone writes a sermon about what I should do and what I shouldn't, but doesn't lose any information about his system and doesn't quote any of the error messages. Sorry, it doesn't work that way.
But here's a bonus for the 1st issue of the year: To run the program just open a terminal and type OpenHantek.
(•‿•)

Ho-Ro, 2020-01-05:
I let this issue rest for a few days, but as almost expected, no further reaction has occurred. I therefore close this issue.
However, I would like to make a few guesses and comments:
The OS seems to be linux, because of talking about .deb, probably a debian or ubuntu based distro.  
> Launching .deb package binary installer fails with a dependency message. It took me a while to realize that you have pre-installation dependency satisfaction instructions. These instructions must [...]

No! As a linux user who installs experimental software from github you should have a minimal understanding how the debian packet management works. If you install the *.deb file correctly, there is no need to install the pre-installation dependencies laboriously by hand, because the debian package management solves this automatically:
Go to the download directory and issue as root: `apt install ./openhantek_..._amd64.deb`. 
Or if you tried to install the package with `dpkg -i openhantek_..._amd64.deb` the subsequent command `apt -f install` fixes all missing dependencies automatically.

> You have a bizarre statement on first page of manual, something about forking, yet don't describe the pre-installation requirements. Replace the bizarre statement with useful information!!!

OH is developed on github and everyone who has an account there should at least know the basic idea behind it - to fork and develop a project - so this statement is not bizarre, but the remark shows the lack of understanding of @www-MKRD-info.

> Not everyone is a developer here, and not everyone has spent a day getting all the pre-requisites!!!

Github **IS** a development platform and the projects here are aimed at developers. Some are really programmers writing code, some propose new applications or algorithms, native speakers do translations, and those who are *just* users can very well help by testing and providing **[meaningful](https://www.chiark.greenend.org.uk/~sgtatham/bugs.html)** error messages.

Martin

closed 2020-01-05 by Ho-Ro

www-MKRD-info, 2020-01-05:
Martin,
I was busy with other things. You were closing this literally as I was writing this.

1. System is Linux Mint 19.3 Cinnamon.

I tried to install this on another computer (same distribution and version), and curiously this time it automatically installed all the dependencies. I will save and follow up on your "fix dependencies automatically" notes next time a .deb package does not do this automatically. Seems to be a difference between my desktop and my laptop Mint 19.3 installs... Unfortunately, I don't have the exact screenshot for the benefit of other users, but it was a red font error on top of .deb package manager saying something like "unable to resolve automatically: libfftw3"

It would be trivial to copy this from your Build instructions page:
**apt install g++ make cmake qttools5-dev qttools5-dev-tools libfftw3-dev binutils-dev libusb-1.0-0-dev libqt5opengl5-dev mesa-common-dev libgl1-mesa-dev libgles2-mesa-dev**

To the package description window, front page of your github page AND the PDF documentation...
Took me a while to find this command in your documentation since I was not following the "build" instructions, thus I was on a goose chase to go find libfftw3 on the Internet, etc for hours...

2, 3. It would be trivial to add this as a first page in your PDF manual. You might as well note to launch the software with (sudo) OpenHantek or follow your method of copying the USB rule.

4. It would be nice to get a Start menu shortcut.

Additional comments:
I needed to actually run **sudo** OpenHantek to be able to download the firmware over USB into the Hantek scope.

Any comments on when the logic analyzer functionality will be added?

www-MKRD-info, 2020-01-05:
I will send other Hantek users towards your software as stock Hantek software is literally useless without the vertical trigger functionality present in stock software, but it would be nice to get logic analyzer functionality soon to completely replace the stock software...

vizkoze, 2020-01-05:
You can use sigrok/pulseview for logic analysis

Ho-Ro, 2020-01-05:
The **sudo** issue looks like you miss a correct udev rule at `/usr/lib/udev/rules.d/60-hantek.rules` (or `/etc/udev/rules.d/60-hantek.rules`).

    # rules for OpenHantek6022 (DSO program) as well as Hankek6022API (python tools)
    
    # Hantek DSO-6022BE, without FW, with FW
    SUBSYSTEM=="usb", ACTION=="add", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="04b4", ATTRS{idProduct}=="6022", TAG+="uaccess", TAG+="udev-acl"
    SUBSYSTEM=="usb", ACTION=="add", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="04b5", ATTRS{idProduct}=="6022", TAG+="uaccess", TAG+="udev-acl"

    # Hantek DSO-6022BL, without FW, with FW
    SUBSYSTEM=="usb", ACTION=="add", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="04b4", ATTRS{idProduct}=="602a", TAG+="uaccess", TAG+="udev-acl"
    SUBSYSTEM=="usb", ACTION=="add", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="04b5", ATTRS{idProduct}=="602a", TAG+="uaccess", TAG+="udev-acl"

    # Voltcraft DSO-2020, without FW (becomes DSO-6022BE when FW is uploaded)
    SUBSYSTEM=="usb", ACTION=="add", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="04b4", ATTRS{idProduct}=="2020", TAG+="uaccess", TAG+="udev-acl"

A menu entry was just added with c0bd578

> It would be trivial to copy this from your Build instructions page:
`apt install g++ make cmake qttools5-dev qttools5-dev-tools libfftw3-dev binutils-dev libusb-1.0-0-dev libqt5opengl5-dev mesa-common-dev libgl1-mesa-dev libgles2-mesa-dev`
To the package description window, front page of your github page AND the PDF documentation...
Took me a while to find this command in your documentation since I was not following the "build" instructions, thus I was on a goose chase to go find libfftw3 on the Internet, etc for hours...

As I wrote there is no need to find `libfftw3-3` - `apt install ./openhantek_..._amd64.deb` will do this for you. The `apt install g++ ...` command installs the packages that are needed for building the software from source, not for installing the *.deb or *.rpm binary package.

> Any comments on when the logic analyzer functionality will be added?

> ...but it would be nice to get logic analyzer functionality soon to completely replace the stock software...

[sigrok](https://sigrok.org/) provides excellent LA functionality. OpenHantek is a DSO only application.
[image]

MarkH1960, 2021-06-16:
Please bear with me I'm a triple virgin. My first Oscilloscope when I'm very new to Ubuntu and this is my very first Git Hub comment question. There's worse to come as I'm also a boomer.

I was bought a Hantek 6022bl by my daughter and I'd rather install it on my Ubuntu laptop than my Windows. That's partially because I'm really enjoying working in Linux and partly because there's a massive difference in value between the two laptops for when I do something stupid and blow it up.

If I've got this right I look for the drivers in the driver updates program and the go into the terminal, I'm presuming at root, type in OpenHantek and follow the directions? Is it really as simple as that sounds? I would genuinely really appreciate any help. Thank you.

Ho-Ro, 2021-06-16:
@MarkH1960: No problem, installation under Ubuntu should be easy-goin'. Download the latest deb package from the [releases](https://github.com/OpenHantek/OpenHantek6022/releases), these are quite stable. Currently it is [3.2.3](https://github.com/OpenHantek/OpenHantek6022/releases/download/3.2.3/openhantek_20210524-ab9a649_amd64.deb). Then open a terminal window, go to the download directory and type `sudo apt install ./openhantek_20210524-ab9a649_amd64.deb`, this will install the required dependencies (e.g. libraries etc.) automatically. You will find an entry under `Applications/Development/Digital Storage Oscilloscope` in the start menu or just start typing OpenH... - it will present the link; if you prefer to start from terminal, just type `OpenHantek`, you should do this as user, not root. That's all...

> My first Oscilloscope ...

Please be careful when measuring high voltage - the scope's input is not isolated from USB that goes directly into your computer that is connected to earth potential of the mains. Never connect the ground of the scope to some voltage because the current can damage the complete system. If you work from battery or wall-wart there's no danger, just keep the measured voltage below 5 volt, for higher voltages up to 50 volt use the x10 position of the provided probes.

> There's worse to come as I'm also a boomer.

I'm even 1 year older :)

MarkH1960, 2021-06-17:
Thank you very much indeed, your help is greatly appreciated.

Ho-Ro, 2021-06-17:
You're welcome, there's also a [QnA section](https://github.com/OpenHantek/OpenHantek6022/discussions) for this project where you can ask, discuss or present something.

[image]

If you want to use the logic analyser of the 6022BL I propose the excellent [sigrok](https://sigrok.org/) project, [BL as LA is supported](https://sigrok.org/wiki/Hantek_6022BL) quite well. I own a 6022BE and use an external "el-ceapo" LA from [Geeetech](http://www.geeetech.com/wiki/index.php/CY7C68013) (or [LCsoft](https://sigrok.org/wiki/Lcsoft_Mini_Board)) for about 10 € / $ with sigrok quite often.

[image]

Just out of curiosity - what's your background and intention to use the scope - ham, audio, digital or something else?

MarkH1960, 2021-06-27:
Sorry I didn't see this latter comment in my newbieness, thank you again.

MarkH1960, 2021-06-27:
I'm actually a chef by trade who did a law degree in my 30s. Neither of which taught me anything about electronics and electrical engineering. I've never been a practical person so I had to promise the wife not to play with high voltages. I worked in a forensic learning disabilities service for about 12 years until I became disabled myself. I can only manage 12.5 hours a week now as service lead for a disability information and advice service. It keeps me doing something which is the important thing. I'm not one of those people who can watch TV all day and with a high level of dementia in my family I want to keep the brain active.