OpenHantek/OpenHantek6022 #55: How to calibrate? Where are cal_zero.py and calibrate.py?

john-iler, 2020-02-01:
I am astounded by how good this software is! I am using OpenHantek6022 3.0.1 FW0204 on Windows 10 with a Hantek 6022BL.

Voltage measurements are off by 10%, so I would like to calibrate the display. I have a modelDSO6022.conf file that I can change values in and see a corresponding effect in the display.

I read the calibration notes in:
https://github.com/Ho-Ro/Hantek6022API/blob/master/README.md#create-calibration-values-for-openhantek

This references both examples/cal_zero.py and examples/calibrate.py. I have searched github in both OpenHantek and OpenHantek6022, and eevblog but cannot find these files anywhere.

Would you please provide a pointer to the python programs, or at least hints on how to determine values for the modelDSO6022.conf file?

Ho-Ro, 2020-02-01:
Short answer - it is in the same repo as the README: https://github.com/Ho-Ro/Hantek6022API/blob/master/examples/calibrate.py 
I don't know If it works with Windows, but I'll give some more info about the config file when I'm back at my laptop tomorrow.

john-iler, 2020-02-01:
Thanks - got it, looks very thorough. I'll take a couple of days to setup Python and test this out. I will report back here on how it goes in Windows.

Ho-Ro, 2020-02-02:
The Hantek 6022 devices have an offset table in the EEPROM as factory setup (at least my 6022BE had). These values are subtracted from the raw 8bit samples that are in "binary offset" format, i.e. 0x80 = 0V, < 0x80 = negative voltages, > 0x80 = positive voltages. For my device I get e.g. 0x86 for the x10 input ranges, with input shorted it measures exactly this raw value.
In addition to the factory setup, which only corrects the raw offset, I have added gain correction values and a more fine-grained (sub-sample) offset correction, which together with oversampling gives an offset correction close to zero.
The config file on the other hand tells the action to be executed to get a correct display, i.e. add -6 to the raw value of 1st channel and then multiply this result by 1.1505.

    ;OpenHantek calibration file for DSO6022
    ;Created by tool 'calibrate.py'
    
    [offset]
    ch0\10mV=-6
    ch1\10mV=3
    ...
    [gain]
    ch0\10mV=1.1505
    ch1\10mV=1.1624
    ...

john-iler, 2020-02-03:
Thank you Ho-Ro for your extremely helpful replies. Things are working pretty well now. My starting point was offsets up to 200mV and voltage measurements 10% low - not too useful! Calibrating with calibrate.py now gives voltage measurements within 2%, and I'm sure most of that error is due to my less than perfect procedures. My offsets are < 15mV. Here is some feedback on changes I needed to make to get there.

Calibrate.py fails to run in Windows 10 with the following error:
"File "C:\Users\johni\AppData\Local\Programs\Python\Python38\lib\site-packages\usb1\__init__.py", line 1354, in kernelDriverActive
    raiseUSBError(result)"

I believe this is because .kernelDriverActive() is not supported in Windows. To correct this I changed LibUsbScope.py line 158:
    old: if self.device_handle.kernelDriverActive(0):
    new: if (os.name != 'nt') and self.device_handle.kernelDriverActive(0):
I also added "import os" at line 11

I then could run "calibrate.py -g -e -c" to generate a "modelDSO6022.conf" file.
This generated reasonable gain values, but the offsets were quite large (e.g. 143). When I ran OpenHantek, there were no traces displayed at all. By trial and error, I had generated a .conf with a set of offsets that worked very well and realized the offset equations appear incorrect (perhaps copied from the eeprom values being written a few lines down?) I made the following changes that generated values that worked very well, and matched my hand generated values.

calibrate.py line 185 (also make similar change for other channel)
    old: config.write( "ch0\\%dmV=%d\n" % ( voltID, 0x80 + offlo1[ gainID ] ) )
    new: config.write( "ch0\\%dmV=%d\n" % ( voltID, -offlo1[ gainID ] ) )

In case you would like to incorporate my modifications, I have attached the 2 modified Python scripts for reference, and the resulting generated .conf file that works well. My changes are marked with "#ji".

[calibrate.py.pdf](https://github.com/OpenHantek/OpenHantek6022/files/4150714/calibrate.py.pdf)
[modelDSO6022.conf.pdf](https://github.com/OpenHantek/OpenHantek6022/files/4150715/modelDSO6022.conf.pdf)
[LibUsbScope.py.pdf](https://github.com/OpenHantek/OpenHantek6022/files/4150717/LibUsbScope.py.pdf)

Thanks again for helping out a newbie and converting my 6022BL from a paperweight to a fun-to-use tool.

closed 2020-02-05 by Ho-Ro