OpenHantek/OpenHantek6022 #56: How to make a logger?

jm-o, 2020-02-04:
I'd like to add a feature to log analyzed data values (rms, vpp...).

I imagined to clone the csv exporter to make a continous logger :
1. Ask for a filename, open a file and write header (columns names),
2. Write into that file the data at fixed interval (1 s for instance) during a specified time or until a mouse clic on the corresponding option menu,
3. Close the file.

I don't understand very well the exporter model.
It seems to process the data, update progress and only at the end ask for a filename to save the processed data.
Is it correct?
So what should be the easiest way to make a logger?

Thanks in advance for your advice.

Ho-Ro, 2020-02-05:
The exporting stuff is my *terra incognita* (hic sunt dracones) - you may know that the original OpenHantek was born about ten years ago to support the *real* Hantek scopes DSO52xx, i.e. the devices with a screen, as well as the *fast* DSO2xxx types, the *small* 6022 were added later. On the other hand my clone OpenHantek6022 took the *original* and cut off everything not related to 52xx and 2xxx about one year ago. Therefore some things may look more complicated than necessary - there's a lot of vintage code inside. But at a quick glimpse I detected this code snippet:
https://github.com/OpenHantek/OpenHantek6022/blob/2ae596f8771e788f401f01eb06026ce6ad682180/openhantek/src/exporting/exporterinterface.h#L41
So I would suppose to create a clone of e.g. exportcsv.cpp and .h, rename it and try to get it called in main.cpp:
https://github.com/OpenHantek/OpenHantek6022/blob/2ae596f8771e788f401f01eb06026ce6ad682180/openhantek/src/main.cpp#L143
Then get the values of interest, e.g. Vrms or frequency and write it out, the next step would be to change the type from SnapshotExport to ContinousExport:
https://github.com/OpenHantek/OpenHantek6022/blob/2ae596f8771e788f401f01eb06026ce6ad682180/openhantek/src/exporting/exportcsv.cpp#L23
One thing that doesn't work with the current program structure is continous sampling of ADC data, because it gets a block of 200000 .. 20000 samples (plus about 1000 samples at the beginning of each block that are unstable and have therefore been dropped), processes them, displays them and get the next block - so no chance to get an uninterrupted data stream. But for sure you can save *low-speed* summary data.

Good luck and thank you for your valuable input, I like your approach.
One additional question, maybe I missed you info, which system do you use?

jm-o, 2020-02-05:
Thank you for your quick reply.

I'll keep you informed of my progress ;-)

I use UBUNTU Linux.

Ho-Ro, 2020-03-22:
That's not exact the solution but maybe a step in this direction: #64

Ho-Ro, 2020-03-26:
I've had a closer look at the source code and still think that a separate logger is a much better solution. The program [capture_6022.py](https://github.com/Ho-Ro/Hantek6022API/blob/master/examples/capture_6022.py) allows to sample both channels with 20..100 kS/s for a long time and save the values in a file. These sampled values can then be analyzed further, e.g. the spectrum of the signals is displayed with [fft_from_capture.py](https://github.com/Ho-Ro/Hantek6022API/blob/master/examples/fft_from_capture.py).

closed 2020-03-26 by Ho-Ro

jm-o, 2020-03-26:
Thank you Ho-Ro!