OpenHantek/OpenHantek6022 #58: Windows 10 pro 64b - Hantek 6022BL Connection Failure

va7ta, 2020-02-13:
Windows 10 pro 64b, version 1809, OS build 17763.1039; Intel Core 2, Duo CPU E6550
OpenHantek-Win-x64-Release-b228/OpenHantek.exe

Device Manager Driver Listing: Hantek6022BL\Hantek6022BL DRIVER 2

Error uploading firmware:
"UploadFirmware: Couldn't open device: Operation not supported or unimplemented on this platform"

Note:
Recent Hantek software installation downloaded 20200212.
Drivers provided within the Hantek install package work OK with the Hantek software version 1.06.

Ho-Ro, 2020-02-14:
It looks as if the manufacturer's driver is still active, with which OpenHantek does not work; the libusb interface is required instead, as also described [there](https://github.com/OpenHantek/OpenHantek6022/blob/master/docs/build.md#windows).
Please provide the information how you did install the OpenHantek6022 software. 
Did you follow the "zadig" way? Did you do it twice?
Did you unplug and plug the device after using it with the Hantek software?

va7ta, 2020-02-15:
Many thanks for the support!!!
I had installed the latest Hantek package which replaced the drivers. Zadig successfully installed the WinUSB drivers and OpenHantek runs now. Not surprised to find the Hantek supplied app will not run with the same drivers as OpenHantek. This issue should be now closed.

closed 2020-02-15 by va7ta

Ho-Ro, 2020-02-15:
Thx for the response! As I'm not using Windows at all, how does the Hantek SW compare to OpenHantek, what is netter, what is missing?

vizkoze, 2020-02-15:
The only "useful" thing is auto detection of timebase and voltage.
But I prefer openhantek6022.

va7ta, 2020-02-15:
My first impression is that I like OpenHantek better than the Hantek app from the factory.

With the Hantek factory app I found the sampling rate jumped from 16 to 48 Mb/s and that when used in the normal mode it would not trigger at the 48 MB/S rate. At slower sampling rates it would only trigger on about 1 out of 5 events. When it did trigger the data burst was offset, far out of view. To find the data the horizontal scale time had to be increased by a factor about 100, then the trace needed to be dragged to center the data for zooming in on it by expanding the view with a faster horizontal scale. An awkward procedure.

With the OpenHantek app when triggered the data was always in view. The sample rate of 24 Mb/s gave a much better view of the data. The cursor measurements work nicely. Unfortunately it also failed to trigger every time. I read the triggering issue is associated with hardware and thus apparently cannot be fixed in software.

Unfortunately the drivers needed for the factory app do not work with OpenHantek. Thus to use the factory provided logic analyzer software one would need to change the drivers. I wonder if there is an open software logic analyzer app that will run with the same drivers as OpenHantek?

Ho-Ro, 2020-02-15:
I can highly recommend [PulseView](https://sigrok.org/doc/pulseview/unstable/manual.html) from the [sigrok project](https://sigrok.org/) that has support for the scope as well as the LA function (compatible with saleae 8 bit devices) of the [6022BL](https://sigrok.org/wiki/Hantek_6022BL) and works also under [Windows](https://sigrok.org/wiki/Windows) with libusb (installed with Zadig).

dyarkovoy, 2020-12-19:
I'm having similar issue on Windows 10 20H2 build 19042.685 with DSO-6022BL and the latest openhantek 3.1.4 (also tried b483, b479 and b461, x64 and x86 variants).
I followed zadig way, tried all combinations of drivers (WinUsb, libusb-win32, libusbK), rebooted mutiple times, tried keeping the UDB plug across installations/reboots, then tried re-plugging. Still I get the error "Connection failed! Couldn't open device: Operation not supported or unimplemented on this platform", but there's a twist: when I select "DSO-6022BL" from the drop-down in openhantek initial window (btw, it shows green), it immediately says "The device is ready for use" and this text stays up for less than a second, then the drop-down automatically changes to "DEMO" and the "Connection failed!" error appears. Now, if I'm fast enough to click OK during these milliseconds when "The device is ready for use" shows, the main window opens and it's almost functional, except that there's no graph and I see "ERR" messages in place of the graph