OpenHantek/OpenHantek6022 #59: Support for other hardware

merbanan, 2020-02-17:
Would patches for other devices be accepted?

https://sigrok.org/wiki/Noname_LHT00SU1
https://sigrok.org/wiki/HT_USBee-AxPro

They have sigrok firmware but no gain stage and only one channel.

Ho-Ro, 2020-02-17:
Yes, all devices with free open source firmware for the analog function are welcome. :)
One channel only and no gain should be handled by the model spec.

merbanan, 2020-02-26:
There is not alot of work done yet, I just wanted to be sure that a contribution would be accepted.

Ho-Ro, 2020-02-26:
Hey, this is a collaborative open source project :) If the core functionality (best support for Hantek6022) is not affected, every **tested** and **actively supported** other device is welcome.

pbarBobo, 2020-03-02:
Hi Guys, is it possible that OpenHantek will work with Hantek IDSO1070 series, this is a wife enabled scope, can also work via USB. 2 channels plus external trigger, 70MHz Bandwidth and 250MSa/s. I recently bought but found original software to be very poor. 

Cheers

Ho-Ro, 2020-03-04:
@pbarBobo I don't want to disappoint you, but OpenHantek6022 only works with the *"low-end"* (Cypress FX2-only) scopes. These devices feature a (nominal) sample rate of 48MS/s maximum (which is what the FX2 could theoretically deliver). Your device scans at 250MS/s, this is only possible with an additional FPGA and buffer RAM, which ultimately results in a different USB protocol that OpenHantek6022 does not support. 
OpenHantek6022 is a spin-off from the original [openhantek project](https://github.com/OpenHantek/openhantek) that works with the *"bigger"* Hanteks. Unfortunately this project is currently [not maintained](https:/github.comOpenHantekopenhantekissues277).

pbarBobo, 2020-03-04:
Hi Ho-Ro, 
                  Thanks for the reply, I had feared that this was the situation. Thank you for taking the time to clarify this. It is a pity as the hardware is quite good but the software is quite bad. I will return this device and purchase a bench scope I think.

Many Thanks

Ho-Ro, 2020-06-28:
> There is not alot of work done yet, I just wanted to be sure that a contribution would be accepted.

I will close here, please reopen if there is something to show.

closed 2020-06-28 by Ho-Ro