OpenHantek/OpenHantek6022 #60: 48MHz mode

merbanan, 2020-02-26:
While working on the FX2 before I always theorized about the 48MS/s mode. As I understood you should be able to sample to the internal FIFO (1-4kib) without drops at that rate. One would then have to transfer the buffer and thus incurring a drop-out. Each buffer would then have to be re-aligned to be able to visualize a nice signal.

Is there support for something like that or does the current handling of samples assume almost a drop free transfer? Or is it the actual FX2 firmware that doesn't wait until all samples are transferred and just overwrite with newer samples in the FX2 FIFO?

Ho-Ro, 2020-02-26:
The FX2 firmware does nothing more than initialize the GPIF, which generates the sample clock for the ADC and transmits the result in 512-byte blocks in USB bulk mode at constant data rate. If the USB bus cannot transfer at this high data rate, it drops one or more of these blocks. This was investigated [in the big eevblog thread some time ago](https:/www.eevblog.comforumtestgearhantek-6022be-20mhz-usb-dsomsg661509). My [firmware](https://github.com/Ho-Ro/Hantek6022API) supports the isochronous mode as alternate 1, but OpenHantek6022 does not (yet) support this mode at the moment.

merbanan, 2020-02-26:
Ok, my memory is getting back to me. The idea was to run a GPIF state machine so it starts filling the quad-buffered 1k FIFO. When the FIFO is full the GPIF stops. The buffers are supposed to be fetched as quickly as possible and when the first (or second) 1k buffer has been received the GPIF is started again via a usb command.

This would fill the USB link with as many continuous 4k buffers as possible. And that is running with a 48MHz sample clock. Using this method periodic signals should be possible to visualize as long as they fit the 4k buffer. Even 2 channels should be possible but with half the buffer size.

And with 30MHz working real-time it should be possible to overrun the 4*1k quad-buffer 2 times (2*1kb) for a possible 6kb large continuous buffer.

Ho-Ro, 2020-02-27:
Your 4K@48MS/s proposal is an interesting approach. As I'm not such a FX2 expert, I need to deep-dive into the TRM when I have a little more time. Please keep in mind that it is important to clock the ADC constantly - due to its [common mode input voltage out of spec](https://github.com/OpenHantek/OpenHantek6022/issues/45#issuecomment-579257234) it needs a long settle time when starting.

merbanan, 2020-02-27:
Ok, even with this limitation it should be possible work around it. Ideally a GPIF state machine should be able to handle the sampling of the buffers on its own. If we start it and it fills 6 1k buffers then we can be sure we get 4 1k blocks of good samples. Then it would be nice if the state machine restart automatically after the last buffer is transferred. Then one could just transfer as fast as possible and still get viable data. Anyway I don't see a way to get the GPIF to operate in that way (yet). One would have to manually start it via the command interface (8150mcu).

For good visualization you want 60fps (or more if you want to apply some filtering) that needs 60*4kb=240kib/s (double that for 2 channels). That is far from what USB can handle but you have to add the control channel traffic to it. But that still sounds doable.

Anyway I just wanted to post the idea. We'll see how far it goes.

Ho-Ro, 2020-06-28:
No progress so far - I will close for the moment - can be reopened in case of new ideas.

closed 2020-06-28 by Ho-Ro

dmantione, 2020-06-30:
Just a small comment: Just migrated from OpenHantek to OpenHantek6022 and I noticed the 48 MHz samplerate is no longer available. As a developer I understand the above discussion, as a user I don't: The oscilloscope resolution is noticeable lower at 30 MHz. I hacked the source code back to 48 MHz and got back the original resolution. Maybe there are USB congestions, but I don't notice them, and can use the scope at its normal resolution again.