OpenHantek/OpenHantek6022 #63: Displayed signal jumps up when sampling with 30Ms/s

pedropedro, 2020-03-14:
Hi Ho-Ro,

Steps to reproduce the behavior:
1. Go to '...'
2. Select '....'
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or a video to help explain your problem.

**Computer environment (please complete the following information):**
 - OpenHantek version: [use always the latest commit, e.g. *release v2.03, commit 1a2b3c4, ...*]
 - OS: [e.g. *linux, macOS, windows, ...*]
 - Distribution, version: [e.g. *debian stable amd64, ...*]
 - Video hardware: [e.g. *Intel Xeon ... Integrated Graphics*]
 - OpenGL version: [e.g. *OpenGL 4.5, Mesa 19.0.3*]
 - Qt version: [e.g. *Qt 5.11.x*]
 - C++ compiler version: [e.g. *gcc 8.3.x*]

**Scope device (please complete the following information):**
 - Device [e.g. *Hantek 6022BE, ...*]
 - Program top line [e.g. *OpenHantek6022 (Build ...) - Device ... (FW ...)*]
 - Input signal: [e.g. *calibration out 1 kHz, sine wave 1 Veff @ 50 Hz, ...*]
 - Probe setting: [e.g. *no probe, probe X1, probe X10, ...*]

**Additional context**
Add any other context about the problem here, e.g. the program output if started in a console like xterm.

Ho-Ro, 2020-03-14:
The offset of the ADC changes when clocked with >= 30 MHz; the EEPROM of the Hantek6022 contains separate offset (as well as gain) [calibration values](https://github.com/Ho-Ro/Hantek6022API/blob/master/docs/README.md#eeprom) for slow sample rates < 30 MS/s and high sample rates >= 30 MS/s. 
Have you [calibrated](https://github.com/Ho-Ro/Hantek6022API/blob/master/README.md#create-calibration-values-for-openhantek) the scope recently? Due to the [brain-dead hardware](https://github.com/OpenHantek/OpenHantek6022/issues/45#issuecomment-579257234) of the Hantek scope the offset can drift over time.
OpenHantek uses the extended calibration values (created by the python calibration program) that allow to drop the offset to a sub-digit level.

Ho-Ro, 2020-03-17:
I cannot follow up unless you provide some more info.

pedropedro, 2020-03-18:
No, I did not calibrate the scope in my (current) Linux VM. I started first with a WinXP VM with the original Hantek software, where I tried the menu function Utility\Calibrate - no feedback from the software. But when I understand it right, that calibration (? when there was one at all) does not matter, OpenHantek in the new Linux VM "flashes" the scope at every boot from scratch.

Ho-Ro, 2020-03-18:
The calibration values for the scope are stored permanently in the small EEPROM, while the firmware is "flashed" into the RAM on every start of the program. There is also the possibility to create a [calibration file](https://github.com/OpenHantek/OpenHantek6022/blob/master/openhantek/src/hantekdso/models/modelDSO6022.conf) that takes priority over the EEPROM calibration.

pedropedro, 2020-03-18:
OK, my mistake.
After running *python3 calibrate.py -ce* the picture got better - sample rate != 30MS/s => rock stable 0V for the calibration signal (half the period ;-), for 30MS/s is the signal jump only ~ 5mV now, means 10x better.
Merci !

closed 2020-03-18 by pedropedro