OpenHantek/OpenHantek6022 #65: Openhantek rollback

JuttDB, 2020-03-18:
hello.
          I am using windows 10 with hantek6022BE. I was able to test the openhantek. now I like to go back to original firmware. can you please guide me to revert the openhantek firmware to original hantek6022BE.

vizkoze, 2020-03-18:
Just unplug the device, stop openhantek and start hantek software. It should load the firmware stored in the official hantek software

JuttDB, 2020-03-18:
i did the same thing but original hantek software searching for the device. it's detecting device any more

vizkoze, 2020-03-18:
Have you tried turning it off and on again? 😉

JuttDB, 2020-03-18:
yes, I removed the USB and connect again. even on the pc side I also did reset.

JuttDB, 2020-03-18:
is there anything that I need to do with Zadig software again.

Ho-Ro, 2020-03-18:
Did you read the Zadig [FAQ](https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it)?

JuttDB, 2020-03-18:
after uninstalling from Universal Serial Bus devices everything is working again.

closed 2020-03-18 by JuttDB

fire1, 2020-03-26:
Change USB port! I notice this but after changing USB port (to use original, saleae logic analyzer and OpenHantek) I use all without any software issues