OpenHantek/OpenHantek6022 #70: DS-6022BL reports firmware version 2.05 with head

Jon-Bright, 2020-04-02:
All of this is with commit `23786d22376c56a3615de3281f299e3fe79a387c`, using a fresh clone and a fresh build.

The current firmware for the DS-6022BL reports itself as 2.05, but the firmware uploader is checking for 2.06.

**To Reproduce**
1. Plug in a DS-6022BL
2. Start OpenHantek

**Expected behavior**
OpenHantek uploads the firmware, proceeds to the main GUI

**Actual behaviour**
OpenHantek finds the device, uploads the firmware, finds the device again, uploads the firmware again, etc.

dmesg shows this:

```
[256150.656145] usb 1-2: new high-speed USB device number 73 using xhci_hcd
[256150.804109] usb 1-2: New USB device found, idVendor=04b4, idProduct=602a, bcdDevice= 0.00
[256150.804112] usb 1-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[256153.201265] usb 1-2: USB disconnect, device number 73
[256154.807960] usb 1-2: new high-speed USB device number 74 using xhci_hcd
[256154.956869] usb 1-2: New USB device found, idVendor=04b5, idProduct=602a, bcdDevice= 2.05
[256154.956874] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[256154.956877] usb 1-2: Product: DSO-6022BL
[256154.956880] usb 1-2: Manufacturer: OpenHantek
[256154.956882] usb 1-2: SerialNumber: C9B3303E4189
```

As can be seen, the firmware identifies itself as 2.05.  Changing the firmware version to 2.05 in modelDSO6022.cpp fixes the problem.  iow, this patch works but is presumably wrong (the firmware itself should be updated):

```
diff --git a/openhantek/src/hantekdso/models/modelDSO6022.cpp b/openhantek/src/hantekdso/models/modelDSO6022.cpp
index 71b60c7..a7ecad2 100644
--- a/openhantek/src/hantekdso/models/modelDSO6022.cpp
+++ b/openhantek/src/hantekdso/models/modelDSO6022.cpp
@@ -145,7 +145,7 @@ void ModelDSO6022BE::applyRequirements(HantekDsoControl *dsoControl) const {
 
 
 // Hantek DSO-6022BL (scope or logic analyzer)
-ModelDSO6022BL::ModelDSO6022BL() : DSOModel(ID, 0x04b5, 0x602a, 0x04b4, 0x602a, 0x0206, "dso6022bl", "DSO-6022BL",
+ModelDSO6022BL::ModelDSO6022BL() : DSOModel(ID, 0x04b5, 0x602a, 0x04b4, 0x602a, 0x0205, "dso6022bl", "DSO-6022BL",
                                             Dso::ControlSpecification(2)) {
     initSpecifications(specification);
 }
```

Ho-Ro, 2020-04-02:
BL upversion was forgotten, fixed with c8a0bff
Thx for reporting

Jon-Bright, 2020-04-02:
Pulled, rebuilt, looks good.  Thanks for the quick fix!