OpenHantek/OpenHantek6022 #73: Timebase wrong, and changes depending on # of channels collected

fovea1959, 2020-04-06:
**Describe the bug**
Depending on the timebase and samplerate, the timebase and frequencies appear to be off, and the error changes based on whether 1 or 2 channels are collected.

**To Reproduce**
- Start OpenHantek6022 (20200406 build 663).
- Connect CH1 to calibration.
- Set calibration out to 60Hz.
- Set timebase to  500us, samplerate to 2MS/s.
- Enable CH1, disable CH2.
- Note that a cycle of the calibration output takes about 3.6 divisions (1.8
ms). Using the measurement tool or markers indicates a cycle time of 1.8ms, which is about 550 Hz. A cycle of a 60hz calibration should take 16.66ms.
[image]
- Enable CH2.
- Note that a cycle of the calibration output now takes about 1.8 divisions (900us). Using the measurement tool or markers indicates a cycle time of 0.9ms, which is about 1100Hz. In this screenshot, Channel2 is the 2nd channel probe picking up the ambient 60Hz (I'm in US), used that as confirmation that the 60Hz calibration output is indeed 60Hz.
[image]

**Expected behavior**
Expect a cycle of 60Hz to take an indicated 16.7ms, and not change if the number of channels collected changes.

**Screenshots**
See above.

**Computer environment (please complete the following information):**
 - OpenHantek version: commit b2c5492.
 - OS: Linux
 - Distribution, version: Ubuntu 18.04
 - Video hardware: NVIDIA Corporation GF119M [Quadro NVS 4200M]
 - OpenGL version: OpenGL 6.3(? how to check, I see /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGL-6.3.so.6.3.0)
 - Qt version: QT 5.9.5
 - C++ compiler version: gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

**Scope device (please complete the following information):**
 - Device: HANTEK 6022be
 - Program top line  OpenHantek6022 (20200406 build 663) (FW0206)
 - Input signal: CH1: calibration out 60HZ, CH2: probe connected to wire I'm holding, picking up ambient 60Hz
 - Probe setting: Probe X1

**Additional context**
No console output is printed.

Ho-Ro, 2020-04-06:
Strange, I cannot reproduce your observation, please see my screen shot with 60 Hz calibration:
[image]
I can change sample rate / time base and everything stays consistent, now CH2 with German 50 Hz from wire in my hand :) :
[image]
But as I'm heading for a new stable release I'll double check this behaviour.

fovea1959, 2020-04-06:
I unplugged the Hantek and plugged into a different USB port (I realized it was on the same bus as another not-in-use HS devices (webcam). Problem went away.

Moved back to same bus as webcam, problem came back.

Don't know if you want to close this, or put something in to detect the situation.

Ho-Ro, 2020-04-06:
In the past we had sometimes the situation that the device detection failed if on a *shared* bus with other HS devices, but I had never a timing failure. The reason could be the bulk transfer with no guaranteed bandwidth, if the bus is congested, packets are dropped. So it is important not to share the bus and don't use hubs.

closed 2020-04-06 by Ho-Ro