OpenHantek/OpenHantek6022 #74: Trigger X position does not stay in place across restarts

fovea1959, 2020-04-06:
**Describe the bug**
The trigger position (yellow pointer at top of grid) does not come back in the same place when OpenHantek is shut down and started up.

**To Reproduce**
1. Open OpenHantek.
2. Position the X trigger marker above the 9th division line.
[image]
3. Close OpenHantek.
4. Look in .config/OpenHantek. See:
```
[scope]
...
trigger\position=0.9
```
5. Start OpenHantek.
6. Notice that the trigger marker is now in a different place (roughly over 0.28)
[image]
7. Close OpenHantek.
8. Notice that the trigger position in .conf files is changed. Suspect that it got read incorrect at startup, since it was displayed in the wrong place at startup. 
```
[scope]
...
trigger\position=0.27999999999999997
```

**Expected behavior**
Expact trigger position to stay in a consistent place, either at 0.5, or the last place moved to.

**Screenshots**
See above.

**Computer environment (please complete the following information):**
 - OpenHantek version: commit b2c5492.
 - OS: Linux
 - Distribution, version: Ubuntu 18.04
 - Video hardware: NVIDIA Corporation GF119M [Quadro NVS 4200M]
 - OpenGL version: OpenGL 6.3(? how to check, I see /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGL-6.3.so.6.3.0)
 - Qt version: QT 5.9.5
 - C++ compiler version: gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

**Scope device (please complete the following information):**
 - Device: HANTEK 6022be
 - Program top line  OpenHantek6022 (20200406 build 663) (FW0206)
 - Input signal: N/A
 - Probe setting: N/A

**Additional context**
None.

Ho-Ro, 2020-04-10:
I cannot reproduce the described behaviour, I have tried to reproduce it with several releases on several machines. The trigger position is saved correctly and also reloaded correctly.
Load and save of settings is straight forward, please reopen if you find out more details.

load-> [dsosettings.cpp#L94](https://github.com/OpenHantek/OpenHantek6022/blob/be8142add1fd499cbfa0d3ac50dd1f0938b75aca/openhantek/src/dsosettings.cpp#L94):

    if (store->contains("position")) scope.trigger.offset = store->value("position").toDouble();

save-> [dsosettings.cpp#L234](https://github.com/OpenHantek/OpenHantek6022/blob/be8142add1fd499cbfa0d3ac50dd1f0938b75aca/openhantek/src/dsosettings.cpp#L234):

    store->setValue("position", scope.trigger.offset );

closed 2020-04-10 by Ho-Ro

Ho-Ro, 2020-06-14:
This issue could happen under special conditions with zoomed display, now fixed with 03816f7.