OpenHantek/OpenHantek6022 #77: Qt has caught an exception thrown

LubuntuSharp, 2020-04-12:
**Describe the bug**
After some time from startup (say half an hour), the program crashes with following message:
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

**To Reproduce**
It happens even if you do nothing , eg you just leave the 1khz calibration going on; I USE GLES OPTION

**Computer environment (please complete the following information):**
 - OpenHantek version: 3.0.3
 - OS: lubuntu i686 16.04
 - Video hardware: [ATI Integrated Graphics*]
 - Qt version: 5.11.1
 - C++ compiler version: gcc 5.4.0

**Scope device (please complete the following information):**
 - Device [Hantek 6022BE
 - Program top line [e.g. *OpenHantek6022 (Build 3.0.3 - Device ... (FW 0205)*]
 - Input signal: calibration out 1 kHz
 - Probe setting: probe X1

**Additional context**
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

LubuntuSharp, 2020-04-13:
Can't be reproduced anymore, After several hours of using never happened again, maybe something related to new libs installation process -> to be closed

closed 2020-04-13 by LubuntuSharp

Ho-Ro, 2020-04-13:
Same at my side, running for >8 hours didn't show any signs of memory corruption. Good that it was a single case.