OpenHantek/OpenHantek6022 #81: Spam in mail notification

Vascom, 2020-04-22:
I see many messages in mail about pre-release unstable.

What is that and when it will stop?

Ho-Ro, 2020-04-22:
It should stop now. I have built and tested a GitHub automation that uses the tool [hub](https://github.com/github/hub), which allows to create binary install packages from selected commits on GitHub. The tool deletes an existing pre-release 'unstable' and replaces it with a new pre-release 'unstable', this triggers the CI builds (appveyor and travis), which then store their artifacts as release assets. This supports my intention to *release early, release often*. I'm also able to provide my home-grown RPi packages for selected commits.

closed 2020-04-22 by Ho-Ro

Vascom, 2020-04-22:
Understand.
OK

I was worried that you could be hacked :)