OpenHantek/OpenHantek6022 #82: Spacebar won't Start/Stop, toggles form elements instead

justlep, 2020-04-24:
In OpenHantek-Win-x64-Release-b259 on Windows 7 Professional 64bit, I am not able to stop/start the scope using the Spacebar key.

1.  I downloaded OpenHantek from:
https://github.com/OpenHantek/OpenHantek6022/releases/download/3.0.4b/OpenHantek-Win-x64-Release-b259.zip
2. I start OpenHantek. Measurement starts fine (Hantek6022BE).
3. I hit Spacebar.
4. I expect the measurement to stop, but it doesn't.
   Instead, the CH1 checkbox in the right panel gets toggled, so measurent continues with CH1  disabled.

Looks like Spacebar only toggles the first (or last used) form element in the right-hand panels.

Tried to set mouse focus elsewhere, but the behavior persists.

I don't know if it's related, but the program console also shows following warning after startup:
```
QObject::connect: Cannot queue arguments of type 'QList<double>'
(Make sure 'QList<double>' is registered using qRegisterMetaType().)
```

As a hotfix, maybe you could add an redundant Shortcut for Start/Stop. Like F4 or so?

Thanks for your help!

closed 2020-04-25 by Ho-Ro

Ho-Ro, 2020-04-25:
Please check the new [binaries](https://ci.appveyor.com/project/Ho-Ro/openhantek6022/builds/32434422/artifacts).

justlep, 2020-04-25:
The <S> key works perfectly. 
Thank a lot for this fast fix!

Ho-Ro, 2020-04-25:
Gerne, Gruß in meine alte Heimat :)

justlep, 2020-04-26:
Vielen Dank & sonnige Grüße zurück :-)

Ho-Ro, 2020-05-08:
Hi, I just checked 3.0.5-rc2 on a Win10 Prof 64 in demo mode (execute it as `OpenHantek -d`) and also the `SPACEBAR` stops/starts the oscilloscope screen, but nevertheless the `S` key works - as well as the `PAUSE` key (if it is still present on your keyboard).

commit 7d5b854f (2020-04-25): fix #82 - WIN: change Start/Stop shortcut from <SPACE> to <S>.
commit e114f90a (2020-04-25): fix #82 - WIN: change Start/Stop shortcut from <SPACE> to <S>.