OpenHantek/OpenHantek6022 #83: Offset on CH1 & CH2

Maddo534, 2020-04-27:
I use a Hantek 6022BL wit OpenHantek on Windows 10.

On both channels the signals are always offset by some mV, even when the Probes are connected to GND.

[image]

Ho-Ro, 2020-04-28:
This is the typical behaviour without offset calibration. I see 500mV/div with x10 setting -> real range 50 mV/div where 1 digit equates about 4 mV. So the displayed offset of 320 / 520 mV is real 32 / 52  mV, which equates 8 / 13 digits (a not so unusual value, caused by the [brain-dead HW design](https://github.com/OpenHantek/OpenHantek6022/issues/45#issuecomment-579257234) of the scope).

Please follow [the hints](https://github.com/OpenHantek/OpenHantek6022/issues/55) for calibration.

Maddo534, 2020-04-29:
Thank you very much for your answer!

I had to include/change specific paths to some files in the python code for it to work.
But except from that the calibration process worked fine and the problem got solved.

Ho-Ro, 2020-04-29:
> I had to include/change specific paths to some files in the python code for it to work.

Please provide your changes - this will help other Win users.

Maddo534, 2020-04-29:
Here are the things I did to get the calibration working.

First of all a little disclaimer for any incorrect terminology or strange solutions.
This was the first time I ever had something to do with python and my knowledge about this consists purely from some googling such as trial and error.

First I downloaded the complete Hantek6022API folder here:
https://github.com/Ho-Ro/Hantek6022API

Also I needed to install libusb1 using pip.

For the calibration I used calibrate_6022.py in the examples folder of the API.

To this file I added the following at the top of the code:
`import sys
sys.path.append('path\to\this\folder\Hantek6022API-master')`

To the file LibUsbScope.py, located in a folder named PyHT6022 I had to add the same code, also right at the top.

When I tried to run everything now there still was an error that libusb-1.0.dll could not be found.
The dll is located where libusb is installed. 
On Windows it should by default be here:
C:\Users\User\AppData\Local\Programs\Python\Python38-32\Lib\site-packages\libusb\_platform\_windows

The solution to this is definitely not a nice one but it worked.
I took the libusb-1.0.dll file from the above path and copied it to
C:\Users\User\AppData\Local\Programs\Python\Python38-32

After I did this I could run calibrate_6022.py from the windows command prompt without any errors.

Ho-Ro, 2020-04-29:
Did you install Hantek6022API? Probably not, I see that this step is missing in the README.md, so I will give you a quick hint (please check if I'm correct, because I do not use Windows and do not have a Windows PC available for testing).
Own Python packages can be installed typically with `python3 setup.py install`, regardless of the OS. This copies the components into `...\site-packages\...` and they become visible for the example programs.
Under Debian Linux, where I develop, this step is hidden in the `Makefile` under the target `install:`, which is also used when building an Debian installation package.
Sorry for the confusion!

Maddo534, 2020-04-29:
Installing solved the issues I had earlier.
But the first time I ran calibrate_6022.py a new error occurred.
The firmware for the scope wasn't where the software searched for it.
After I shifted the firmware folder to the directory the software expected it to be everything worked again.

But I'm not certain if this is a general problem if you do this on Windows or if it roots in my lack of experience with python and thus an me-made error with the installation, some directory or whatever.

A definitive thing to say is, that when you install Hantek6022API less tweaks are necessary.

Ho-Ro, 2020-04-30:
I opened a new [issue #8](Ho-Ro/Hantek6022API#8) in the [right repo](https://github.com/Ho-Ro/Hantek6022API), to concentrate all discussion about the python tools under Windows.

closed 2020-04-30 by Ho-Ro