OpenHantek/OpenHantek6022 #84: Issue with ARMHF deb installer package - Raspberry Pi

DarS007, 2020-04-28:
Hello,
Let me report some pecularities with the precompiled Paspberry Pi deb package. It looks that the packade does NOT INCLUDE any resource files like firmware or application images/pictures. When it is installed on RPi, only the following files are placed: 
``` 
> sudo apt install ./openhantek_20200414-52c9236_armhf.deb
The following packages will be installed:    libqt5opengl5 openhantek

> dpkg-query -L openhantek
/lib/udev/rules.d/60-hantek.rules
/usr/bin/OpenHantek
/usr/share/applications/OpenHantek.desktop
/usr/share/doc/openhantek/CHANGELOG
/usr/share/doc/openhantek/LICENSE
/usr/share/doc/openhantek/OpenHantek6022_User_Manual.pdf
/usr/share/icons/hicolor/48x48/apps/OpenHantek.png
/usr/share/icons/hicolor/scalable/apps/OpenHantek.svg
```
Therefore any attempt to run such application ends with the error:
```
> OpenHantek
QFileDevice::seek: IODevice is not open
QIODevice::read (QFile, ":/firmware/dso6022be-firmware.hex"): device not open
EOF without EOF record!
```
[ btw - the error description is misleading. It is not the problem with opening the DEVICE (in a sense of Hantek DSO hardware), but with opening the missing FIRMWARE FILE ]

Because the resource files are missing, the application cannot find the firmware and upload it to the hardware.

A quick check confirmed that 'local compilation' can help. The locally compiled OpenHantek6022 runs fine, without any problems (great piece of software!, btw). No error messages of any kind on my RPi3.

And, of course, the Hantek 6022BE hw with properly downloaded firmware (via the correct OpenHantek6022) works fine with the malfunctioning app from .deb package. But then it complains about other missing resources:
```
Cannot open file ':/images/pause.svg', because: No such file or directory
Cannot open file ':/images/play.svg', because: No such file or directory
Cannot open file ':/images/digitalphosphor.svg', because: No such file or directory
Cannot open file ':/images/histogram.svg', because: No such file or directory
Cannot open file ':/images/zoom.svg', because: No such file or directory
Cannot open file ':/images/measure.svg', because: No such file or directory
Cannot open file ':/images/OpenHantek.svg', because: No such file or directory
```

Deb package is a convenient way of installing software, in particular for more novice users (perhaps many RPi users are in this category). Can you make the needed changes to make deb working flawlessly on RPi? Or advise, where to manually place the firmware directory so the users could perform this step themselves. I tried to guess the needed location, but gave up after multiple failures.

If the fix can be done by manually placing firmwares in the needed directory (where?), I can later update the installation instructions on OpenHantek6022 page.

Regards,
-DarS

Ho-Ro, 2020-04-28:
Hi, I just did check the RPi installation of the same package downloaded from github (after purging the old openhantek package):

    Selecting previously unselected package openhantek.
    (Reading database ... 160929 files and directories currently installed.)
    Preparing to unpack .../openhantek_20200414-52c9236_armhf.deb ...
    Unpacking openhantek (20200414-52c9236) ...
    Setting up openhantek (20200414-52c9236) ...
    Processing triggers for mime-support (3.62) ...
    Processing triggers for hicolor-icon-theme (0.17-2) ...
    Processing triggers for gnome-menus (3.31.4-3) ...
    Processing triggers for desktop-file-utils (0.23-4) ...

`$ OpenHantek`

    libEGL warning: DRI2: failed to authenticate
    qt5ct: using qt5ct plugin
    qt5ct: D-Bus global menu: no

It shows the three warnings above and the program starts and runs without further issues.

> It looks that the packade does NOT INCLUDE any resource files like firmware or application images/pictures. 

You're right, there are no separate resource files, because the resources (translation, firmware and images) are compiled together with the code into the binary `OpenHantek`. If you open the binary with an editor, e.g. `mcedit` from *midnight commander* you can search for the *.svg names (e.g. search for the string `name="pause.svg"`) and you will see the readable svg code. Same goes for the firmware, search for `:00000001FF` and it will show you the intel hex code. If you have an already **correct** intalled deb you can even download the *.tar.gz file (that includes only the binary) and run it from where you want. 

So I do not understand what went wrong with your install. What does `which OpenHantek` say? Do you have any old binary installed?

I build RPi packages for the releases (semiautomatically) on one of my RPi3 and upload them as assets to github, the intel packages for Linux, Mac and Win are built automatically  by the CI provider `travis-ci.org` and `ci.appveyor.com`. The `*_amd64.deb` is tested thoroughly on my Debian system, while I install the RPi packages only from time to time.

This is the `INFO` section of the deb, the dependencies are automatically detected during build and missing packages should also be installed automatically if not available on the system (as libqt5opengl5 in your case).

    new Debian package, version 2.0.
    size 4127022 bytes: control archive=761 bytes.
        509 bytes,    10 lines      control              
        593 bytes,     8 lines      md5sums              
    Architecture: armhf
    Depends: libc6 (>= 2.4), libfftw3-double3 (>= 3.3.5), libgcc1 (>= 1:3.5), libgl1, 
    libqt5core5a (>= 5.6.0~beta), libqt5gui5 (>= 5.6.0~beta), libqt5opengl5 (>= 5.0.2), 
    libqt5printsupport5 (>= 5.2.0), libqt5widgets5 (>= 5.4.0), libstdc++6 (>= 5), libusb-1.0-0 
    (>= 2:1.0.16)
    Description: Digital oscilloscope software for Hantek DSO6022 USB hardware
    Maintainer: contact@openhantek.org
    Package: openhantek
    Priority: optional
    Section: electronics
    Version: 20200414-52c9236
    Installed-Size: 5076

There are some other RPi users, but no one saw this issue before

The statistic for the latest stable release shows some RPi downloads:

    OpenHantek/OpenHantek6022/releases/latest                       updated   dl
    openhantek-20200414_52c9236-1.armv7l.rpm           2020-04-14T20:02:07Z   20
    openhantek-20200414_52c9236-1.x86_64.rpm           2020-04-14T06:53:49Z   41
    OpenHantek-Win-x64-Release-b259.zip                2020-04-14T06:56:44Z  183
    OpenHantek-Win-x86-Release-b259.zip                2020-04-14T06:56:46Z   54
    openhantek_20200414-52c9236-1_armv7l.tar.gz        2020-04-14T20:02:11Z    7
    openhantek_20200414-52c9236-1_osx_x86_64.dmg       2020-04-14T08:06:14Z   47
    openhantek_20200414-52c9236-1_osx_x86_64.tar.gz    2020-04-14T08:06:16Z   10
    openhantek_20200414-52c9236-1_x86_64.tar.gz        2020-04-14T06:53:50Z   21
    openhantek_20200414-52c9236_amd64.deb              2020-04-14T06:53:48Z  123
    openhantek_20200414-52c9236_armhf.deb              2020-04-14T20:02:11Z   20

and the previous 3.0.3 has even more:

    OpenHantek/OpenHantek6022/releases/tags/3.0.3                   updated   dl
    ...
    openhantek_20200301-febf065_armhf.deb              2020-03-17T21:07:01Z   49

So please purge the old *.deb, reboot the RPi, load a fresh *.deb (maybe you try also the *unstable* release, it is quite stable at the moment as I want to release it as 3.0.5 the next days.)

HTH, Martin

DarS007, 2020-04-28:
Thanks for the details! Great knowledge. It's clear who is THE developer :-) I'm glad I can learn from you.
>  the resources (translation, firmware and images) are compiled together with the code into the binary OpenHantek.

Yeah, I did not realize the binary is so monolithic. So I was wrong assuming that the app is using external resource files (and the missing firmware file caused the above mentioned error). I will then troubleshoot further tomorrow. 

Knowing now how all pieces are packed together, I start suspecting that the content unpacking malfunctioned for some reason. Only now I realized that I'm doing my debugging on a pretty old Raspbian... (some technical compatibility reasons kept me from installing the latest release). I will check the details and perform few more tests. Including the play with the latest Raspbian - which I wanted to avoid :-)

I will update afterwards.

Best regards,
-DarS

Ho-Ro, 2020-04-29:
> A quick check confirmed that 'local compilation' can help. The locally compiled OpenHantek6022 runs fine, without any problems (great piece of software!, btw). No error messages of any kind on my RPi3.

You can also build your own *.deb in the `build` directory:

    sudo rm -f packages/*
    make -j4 && sudo make package && sudo chown -R <youruser>:<yourgroup> packages/*

Important: it's `make package`, but the target directory is `packages`. You should end up with three packages in `packages`, deb, rpm and tar.gz, the latter contains only the binary.

DarS007, 2020-04-29:
OK, my tests are over now. 

Indeed, I initially run OpenHantek6022 on **older** (but freshly updated) Raspbian Stretch. This is where I encountered the issue with the app not being able to flash the firmware. I spent several hours today to perform OS version upgrade from Stretch to Buster. I didn't want to go for a plain vanilla Buster SD card because I wanted to check any potential side effects of the version upgrade. Got several nasty surprises in the meantime. Like with the 8GB SD card that run out of space:
```
dpkg: unrecoverable fatal error, aborting:
 failed to write status database record about 'libwavpack1:armhf' to '/var/lib/dpkg/status': No space left on device
E: Sub-process /usr/bin/dpkg returned an error code (2)
```
I had to manually free the space and finish/clean the upgrade. Battle as usual... Overall, the refreshed RPi works quite well, the 5" Chinese LCD still displays the desktop (I was worrying about LCD drivers compatibility with Buster). External WiFi dongle failed to work, so I temporarily use the Eth cable.

Once the OS upgrade was done, I again downloaded the OH6022 deb package from your GitHub repo and installed it on the refreshed RPi. I'm glad to announce that **it works now**. Please close this case. The main cause of the initial bad OpenHantek6022 behavior on Raspbian Stretch is unknown to me, but I'm glad that the OS upgrade fixed it.

closed 2020-04-29 by DarS007

Ho-Ro, 2020-04-29:
Thx for reporting

DarS007, 2020-04-29:
And big thanks to YOU. Not only for your help here (incl. the recipe for .deb package creation), but for involvement and maintenance of this branch. OpenHantek6022 is not only a great piece of software, but together with cheap Hantek 6022 DSO and Raspberry Pi forms the ultimate hobbyist tool.