OpenHantek/OpenHantek6022 #88: Software not exiting properly.

freebird61, 2020-05-08:
Hi

I've been using OpenHantek6022 for a few months (tried various versions from 3.0.1 to 3.1.0 - binaries and local build) but always have the same issue.  On first connection of the 6022BE (i.e. the device requires initializing and firmware upload), quitting OpenHantek6022 closes the window but doesn't exit the process - I have to kill it (or ctrl-c if run from the command line) - no error messages produced.  Subsequent launches of OpenHantek6022 work fine.

I've discovered that if I re-initialize the device after firmware upload, OpenHantek6022 exits OK ... my 3.1.0 hack attached.

Having googled and not found anyone else with the same problem, I'm beginning to suspect my workstation build (Debian 9, xfce) is at fault although the unmaintained OpenHantek software doesn't exhibit the issue.  Any ideas would be appreciated.    

Nige

[OpenHantek6022.3.1.0_main.patch.txt](https://github.com/OpenHantek/OpenHantek6022/files/4598684/OpenHantek6022.3.1.0_main.patch.txt)

freebird61, 2020-05-08:
A bit more info.
Save settings dsoSettings->save() in MainWindow::closeEvent completes OK then it hangs on a futex FUTEX_WAIT_PRIVATE.  Strace output ...

rename("/home/nige/.config/OpenHantek/OpenHantek6022.conf.M16484", "/home/nige/.config/OpenHantek/OpenHantek6022.conf") = 0
stat("/home/nige/.config/OpenHantek/OpenHantek6022.conf", {st_mode=S_IFREG|0644, st_size=3367, ...}) = 0
stat("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3687, ...}) = 0
stat("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3687, ...}) = 0
stat("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3687, ...}) = 0
close(25)                               = 0
unlink("/home/nige/.config/OpenHantek/OpenHantek6022.conf.lock") = 0
stat("/home/nige/.config/OpenHantek.conf", {st_mode=S_IFREG|0644, st_size=0, ...}) = 0
open("/home/nige/.config/OpenHantek.conf", O_RDONLY|O_CLOEXEC) = 25
fcntl(25, F_SETFD, FD_CLOEXEC)          = 0
fstat(25, {st_mode=S_IFREG|0644, st_size=0, ...}) = 0
stat("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3687, ...}) = 0
stat("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3687, ...}) = 0
stat("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3687, ...}) = 0
close(25)                               = 0
stat("/etc/xdg/OpenHantek/OpenHantek6022.conf", {st_mode=S_IFREG|0644, st_size=0, ...}) = 0
open("/etc/xdg/OpenHantek/OpenHantek6022.conf", O_RDONLY|O_CLOEXEC) = 25
fcntl(25, F_SETFD, FD_CLOEXEC)          = 0
fstat(25, {st_mode=S_IFREG|0644, st_size=0, ...}) = 0
stat("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3687, ...}) = 0
stat("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3687, ...}) = 0
stat("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3687, ...}) = 0
close(25)                               = 0
stat("/etc/xdg/OpenHantek.conf", {st_mode=S_IFREG|0644, st_size=0, ...}) = 0
open("/etc/xdg/OpenHantek.conf", O_RDONLY|O_CLOEXEC) = 25
fcntl(25, F_SETFD, FD_CLOEXEC)          = 0
fstat(25, {st_mode=S_IFREG|0644, st_size=0, ...}) = 0
stat("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3687, ...}) = 0
stat("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3687, ...}) = 0
stat("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3687, ...}) = 0
close(25)                               = 0
write(3, "\1\0\0\0\0\0\0\0", 8)         = 8
poll([{fd=4, events=POLLIN|POLLOUT}], 1, -1) = 1 ([{fd=4, revents=POLLOUT}])
writev(4, [{iov_base="\24\0\6\0n\0`\4[\1\0\0\0\0\0\0\0\0\0\0\0\4\0\0", iov_len=24}], 1) = 24
futex(0x7fff1fab4294, FUTEX_WAIT_PRIVATE, 1, NULL) = 0
futex(0x55bed84cf368, FUTEX_WAKE_PRIVATE, 1) = 0
poll([{fd=4, events=POLLIN|POLLOUT}], 1, -1) = 1 ([{fd=4, revents=POLLOUT}])
writev(4, [{iov_base="\24\0\6\0n\0`\4@\1\0\0\4\0\0\0\0\0\0\0\0\4\0\0", iov_len=24}], 1) = 24
futex(0x7fff1fab4244, FUTEX_WAIT_PRIVATE, 1, NULL) = -1 EAGAIN (Resource temporarily unavailable)
futex(0x55bed84cf368, FUTEX_WAKE_PRIVATE, 1) = 0
write(3, "\1\0\0\0\0\0\0\0", 8)         = 8
write(3, "\1\0\0\0\0\0\0\0", 8)         = 8
write(3, "\1\0\0\0\0\0\0\0", 8)         = 8
write(3, "\1\0\0\0\0\0\0\0", 8)         = 8
write(24, "\1\0\0\0\0\0\0\0", 8)        = 8
futex(0x55bed87c9c2c, FUTEX_WAIT_BITSET_PRIVATE, 1, {tv_sec=22705, tv_nsec=941757163}, 0xffffffff) = 0
futex(0x55bed87c9c00, FUTEX_WAKE_PRIVATE, 1) = 0
fstat(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 1), ...}) = 0
write(23, "\1\0\0\0\0\0\0\0", 8)        = 8
futex(0x55bed875f3dc, FUTEX_WAIT_BITSET_PRIVATE, 1, {tv_sec=22705, tv_nsec=950228270}, 0xffffffff) = 0
futex(0x55bed875f3b0, FUTEX_WAKE_PRIVATE, 1) = 0
ioctl(22, USBDEVFS_RELEASEINTERFACE, 0x7fff1fab498c) = 0
write(17, "\1", 1)                      = 1
close(22)                               = 0
read(16, "\1", 1)                       = 1
poll([{fd=16, events=POLLIN}, {fd=18, events=POLLIN}], 2, 0) = 0 (Timeout)
futex(0x7f547c001810, FUTEX_WAIT_PRIVATE, 2, NULL

Ho-Ro, 2020-05-09:
Strange... I will try to dig into, but as you wrote I never got a feedback about this kind of issue.
Does it happen also if you build with clang instead of gcc? In directory `build` do:

    CC=/usr/bin/clang CXX=/usr/bin/clang++ cmake ..
    make

My main system is a laptop with debian 10 (buster), that was updated over the years from jessie -> stretch -> buster, but I also run the prog on a fresh installed buster on another laptop (the hdd crashed recently). I do this gcc <-> clang comparison from time to time without detecting different behaviour.
Quick googling `FUTEX_WAIT_PRIVATE` mentioned some hangs of different programs. (?)

Martin

freebird61, 2020-05-09:
Same result using clang (once I got cmake to use it by removing the build dir).

I'm wondering if other users may have the same issue without realizing ... I only noticed because xfce was taking a long time to close down.  Although the erroneous process is parented by the xfce session, it can't kill it (probably because it no longer has a window associated with it) and when the xfce session eventually does end (20 secs or so), the process becomes parented by init.  Also, if I leave a hung process running, subsequent OpenHantek6022 sessions run fine and exit cleanly.  Could it be an xfce anomaly that can't get the app to quit (I don't suppose many users will be using an xfce desktop) whilst gnome, kde, etc don't show the issue up.

What I don't understand is why restarting the USB device should enable whichever thread is hanging to exit properly.

BTW I should have googled FUTEX_WAIT_PRIVATE (my bad) - it just indicates that the parent is waiting for a child thread to exit.  If I'd dug a little deeper with strace and looked for threads, I'd have seen ...

strace -f -p 20106
strace: Process 20106 attached with 14 threads
[pid 20119] restart_syscall(<... resuming interrupted restart_syscall ...> <unfinished ...>
[pid 20117] restart_syscall(<... resuming interrupted restart_syscall ...> <unfinished ...>
[pid 20116] restart_syscall(<... resuming interrupted restart_syscall ...> <unfinished ...>
[pid 20118] restart_syscall(<... resuming interrupted restart_syscall ...> <unfinished ...>
[pid 20115] futex(0x2410454, FUTEX_WAIT_PRIVATE, 165, NULL <unfinished ...>
[pid 20114] futex(0x24102ec, FUTEX_WAIT_PRIVATE, 165, NULL <unfinished ...>
[pid 20112] futex(0x241001c, FUTEX_WAIT_PRIVATE, 165, NULL <unfinished ...>
[pid 20111] futex(0x240feb4, FUTEX_WAIT_PRIVATE, 165, NULL <unfinished ...>
[pid 20113] futex(0x2410184, FUTEX_WAIT_PRIVATE, 165, NULL <unfinished ...>
[pid 20110] futex(0x240fd4c, FUTEX_WAIT_PRIVATE, 165, NULL <unfinished ...>
[pid 20109] futex(0x240fbe4, FUTEX_WAIT_PRIVATE, 165, NULL <unfinished ...>
[pid 20108] futex(0x240fa7c, FUTEX_WAIT_PRIVATE, 165, NULL <unfinished ...>
[pid 20107] restart_syscall(<... resuming interrupted poll ...> <unfinished ...>
[pid 20106] futex(0x7f3008001810, FUTEX_WAIT_PRIVATE, 2, NULL

Nige

Ho-Ro, 2020-05-09:
Sorry, forgot to mention that I use kde and sometimes lxqt, so maybe the Qt program OH6022 feels better than under GTK?

freebird61, 2020-05-09:
Definitely looking like a weird QT issue.  Commenting out the "start scope button click " hack in selectsupporteddevice.cpp and selecting close in the "select device" dialog box exhibits the same issue.  Any suggestions how to debug further would be useful as I have zero experience with QT coding.

Nige

Ho-Ro, 2020-05-09:
Does it happen also in demo mode? This could safely exclude USB, where I was faced with some problems especially with different OSs (Win, BSD, Mac) due to slightly different implementation.
*OpenHantek6022* as a clone of *openhantek* carries a ten year code heritage, I already did two or three times a big *industrial cleaning*. 
My preferred *debugging* is done by spreading `printf(...)`, `std::cout << ...` or `qDebug() << ...` statements, e.g. to create the stacked output `OpenHantek6022 stopped after .. s` at the end of `main.cpp` to catch a dirty shutdown of the threads. During real debugging one `unitbuf` line was added before the real text output.

    std::cout << std::unitbuf; // enable automatic flushing
    std::cout << "OpenHantek6022 ";
    ...

freebird61, 2020-05-09:
Demo mode exits cleanly which is strange as closing the dialog box is much earlier in the code and I would have expected fewer threads to have started.  The demo mode must be causing the guilty thread to exit whereas the close dialog box doesn't. Am I correct in assuming Qt handles the close as I can't see any reference to handle a close button click in the code?

Ho-Ro, 2020-05-09:
Qt creates a `QApplication`, some threads and finally the `MainWindow`, that is shown. As soon as the application is executed the main program loop is active and handles all user interactions. Closing the main window deletes all child widgets and finally stops the execution of the application, `openHantekApplication.exec()` returns and the code after /// Clean Up /// is executed.

    QApplication openHantekApplication( argc, argv );
    ...
    MainWindow openHantekMainWindow( &dsoControl, &settings, &exportRegistry );
    ...
    openHantekMainWindow.show();
    ...
    int res = openHantekApplication.exec();
    
    //////// Clean up ////////
    ...

If the program hangs, does the program reach the `// Clean up //` part? Maybe you could activate the automatic flushing as mentioned in my comment earlier tonite to see if something is written to `std::cout`.

Please provide also the output of `apt-cache policy libqt5core5a libusb-1.0-0`, these are my versions:

    libqt5core5a:
      Installiert:           5.11.3+dfsg1-1+deb10u3
    ...
    libusb-1.0-0:
      Installiert:           2:1.0.22-2
    ...

I will deep-dive tomorrow into the libusb handling, because the biggest difference between normal mode and demo mode is the USB communication with a lot of dynamic buffering and linked lists as C code, where you can easily shoot yourself in the foot.

freebird61, 2020-05-10:
Got the issue down to the final usb statement in main - libusb_exit( context ) doesn't return.

libqt5core5a:
  Installed: 5.7.1+dfsg-3+deb9u2

libusb-1.0-0:
  Installed: 2:1.0.21-1

freebird61, 2020-05-10:
Hi Martin, I've sorted it.  Pulled and built latest libusb https://github.com/libusb/libusb - works fine now.  Will use a private environment for OH6022 so I'm not messing with the distro.  Apologies for wasting your time. Nige

freebird61, 2020-05-10:
For completeness, running libusb-1.0-0 with LIBUSB_DEBUG=5, libusb_exit was hanging before destroying devices

  [ 5.983884] [00007a73] libusb: debug [libusb_exit]  
  [ 5.983888] [00007a73] libusb: debug [libusb_exit] destroying default context
  [ 5.983895] [00007a73] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
  [ 5.983899] [00007a73] libusb: debug [handle_events] poll fds modified, reallocating
  [ 5.983906] [00007a73] libusb: debug [handle_events] poll() 2 fds with timeout in 0ms
  [ 5.983912] [00007a73] libusb: debug [handle_events] poll() returned 0
Hanging here
  [ 5.983917] [00007a73] libusb: debug [libusb_unref_device] destroy device 3.1
  [ 5.983921] [00007a73] libusb: debug [libusb_unref_device] destroy device 10.1
  [ 5.983926] [00007a73] libusb: debug [libusb_unref_device] destroy device 9.1
  ...

Thanks for your help.
Nige

freebird61, 2020-05-10:
PS On the upside, I've learnt a lot about USB devices :)

Ho-Ro, 2020-05-10:
I'm glad that the issue is solved for you, but I'm going to do a refactoring / renaming in the USB code area - there are many ambiguities in the variable names, e.g. `libusb_device *device` vs. `USBDevice *device`, where the latter should better be called `scopeDevice` and so on.

closed 2020-05-10 by Ho-Ro