OpenHantek/OpenHantek6022 #89: Option to substitute PE1 for PE3 in AC coupling mod

tnm23, 2020-05-11:
It appears I have an early version of the Hantek 6022BL, or some other variant - the circuit board does not have jumper JP2, which makes the AC modification rather challenging!  The pins on the surface mount chip are very close together and I don't want to risk trying to solder bridge wires directly to them; however, pins PE0 and PE1 do still have tracks connected between them and the edge connector at the rear of the PCB, and it's just about possible to piggy-back bridge wires for the opto-switches onto these tracks (this will presumably make the rear edge connector unusable, but I don't expect I'll ever need it).  Can you please thus add a compilation option to use pin PE1 in place of PE3 for the AC-coupling modification?  These are the only two port pins that are accessible for hobbyist-level soldering on this particular version of the board.

[image]

Ho-Ro, 2020-05-11:
Hi,
my firmware for the 6022BL lives [there](https://github.com/Ho-Ro/Hantek6022API/tree/master/PyHT6022/Firmware/DSO6022BL), but it shares most code with  [`DSO6022BE/scope6022.inc`](https://github.com/Ho-Ro/Hantek6022API/blob/master/PyHT6022/Firmware/DSO6022BE/scope6022.inc), e.g. [`static void set_coupling(BYTE coupling_cfg)`](https://github.com/Ho-Ro/Hantek6022API/blob/e4976922db9121780bc13f25974e4f025e0acbff/PyHT6022/Firmware/DSO6022BE/scope6022.inc#L99)
I've hacked a quick *PE1 instead of PE3* version for you that you can put in the firmware directory `openhantek/res/firmware` as `dso6022bl-firmware.hex`.
If you're on a Linux system [it's easy to build your own firmware](https://github.com/Ho-Ro/Hantek6022API#now-with-linux-support), YMMV for Windows.

HTH Martin

[dso6022bl-firmware_PE1_instead_of_PE3.hex.gz](https://github.com/OpenHantek/OpenHantek6022/files/4611288/dso6022bl-firmware_PE1_instead_of_PE3.hex.gz)

tnm23, 2020-05-11:
Thanks very much!  Busy with something else tonight, but I might be able to give it a whirl tomorrow.

Ho-Ro, 2020-05-18:
As I didn't receive any complaints since one week I suppose the solution works for you.

closed 2020-05-18 by Ho-Ro

tnm23, 2020-06-11:
Works like a charm.  My apologies for the excessive delay in confirmation; life is somewhat complicated right now.

KickWilstra, 2020-10-04:
Is it possible to explain in detail about where you connected the bridge wires? I also have this early pcb board version but it's hard to make up on the photo which layers you used. I add a photo of my pcb which can be used to point out the layers.
[image]

Thanks in advance

Ho-Ro, 2020-10-04:
If you look at the [FX2 datasheet](https://www.cypress.com/file/138911/download) you will be able to identify the relevant processor pins on page 21. You can solder your wires to the two vias that I marked in the detail picture below.
[image]

tnm23, 2020-10-04:
Personally, I found the vias were a little too near the IC pins to be comfortable soldering to with my not-exceptionally-fine-tipped soldering iron without risking the whole thing blobbing up with solder, so instead I carefully scraped the enamel off a short patch of each of the two tracks the vias led to at the bottom of the image (making sure these two patches were staggered in position so the solder couldn't run together between them), and soldered the trigger wires directly onto those.  I've highlighted them in the original photo, and done my best at a close-up of the solder joints.  Unfortunately lighting conditions weren't ideal for macro photography, and I used a fair bit of hot-melt glue to secure it all afterwards so the wires couldn't accidentally peel the track from the board, which also makes it a bit hard to see detail.

[image]

[image]

tnm23, 2020-10-04:
Here's also some close-ups of my particular variation on installing the opto-switches.  Again, sorry for the very poor lighting.

[image]
[image]

Ho-Ro, 2020-10-05:
I can see that you use fairly thick wires - for years I have had the best results when using Teflon insulated wire-wrap wires. Cut them to the required length, dip the end into a drop of solder at the tip of the soldering iron for 5..10 seconds (increase the soldering temperature to about 350°C). The Teflon shrinks by about one millimeter at the end and the wire is tinned. This end can then easily be soldered directly onto the pins of (normal sized) ICs (e.g. like the FX2 processor of the Hantek), do not use too much solder. It helps to use e.g. a magnifying glass or (my solution) cheap reading glasses with 3 or 3.5 diopters, which you can find in the drugstore.

KickWilstra, 2020-10-05:
Thanks for the fast response and tips. It helped me understanding the pcb. I followed the pe0 and pe1 tracks and came out at the pcb connector next to the usb connector. The resistors are connected to the interface tracks on the backside of the pcb. Do you think it's also possible to connect the bridge wires at that point? the only difference is tha c80 and C81 are connected to the resistors (see photo).

[image]

tnm23, 2020-10-05:
If you're going to solder that far towards the rear of the board, you might just as well solder directly onto those nice, big pads on the edge connector itself - the connector will probably be unusable after this mod anyway, since we're repurposing some of its data lines for something else.  I'd be wary of making the bridge wires any longer than they need to be, however, because at the end of the day we're trying to build a scientific instrument and we want to pick up as little interference as possible.  It's bad enough that these boards were clearly designed to have metal screening boxes around the input stages, but never had them installed.

As for the thickness of the wire, thanks for the tips, but good old solid-core bell wire was all I had lying around, and I didn't have any very thin solder either - I'm rather old-school, I still mostly use strip-board and through-hole components wherever they wouldn't be too bulky.  I find one can really quite often make a strip-board design very compact indeed if one takes the time to refine the layout through two or three iterations before building it, and it rather encourages one to think a problem through carefully to come up with a really elegant design with a minimum of components, too - but I digress.  I suppose I could have ordered a reel of very fine rework wire, or whatever it's called, when I ordered the opto-switches, but I didn't want to buy a whole reel of specialist wire that I'd likely never use again for years - lavishing money on components for this scope rather defeats the purpose of buying such a budget scope in the first place!

prokrypt, 2021-01-13:
Gosh, I tried to follow the original instructions as much as possible, even drilling holes in the PCB ;)
[image]
[image]
[image]

But it works!