OpenHantek/OpenHantek6022 #91: Russian translation

spam-receiver, 2020-05-12:
<location filename="../src/mainwindow.cpp" line="80"/>
        <source>Demo Mode</source>
        <translation>демонстрационного режима</translation>
Change to
        <translation>Демо Режим</translation>

        <location filename="../src/selectdevice/selectsupporteddevice.cpp" line="73"/>
        <source>&lt;p&gt;Please make sure you have installed the windows usb driver correctly&lt;/p&gt;</source>
        <translation>&lt;p&gt;Убедитесь в корректно установленных USB драйверах для Windows&lt;/p&gt;</translation>
Change to
         <translation>&lt;p&gt;Убедитесь в корректно установленных USB драйверах для Windows.&lt;/p&gt;</translation>

        <location filename="../src/selectdevice/selectsupporteddevice.cpp" line="86"/>
        <source>&lt;hr/&gt;&lt;p&gt;Even without a device you can explore the program&apos;s function. Just press the &lt;b&gt;Demo Mode&lt;/b&gt; button below.&lt;/p&gt;</source>
        <translation>&lt;hr/&gt;&lt;p&gt;Вы можете изучить функции программы без устройства, просто нажмите кнопку &lt;b&gt;демонстрационного режима&lt;/b&gt; кнопки.&lt;/p&gt;</translation>
Change to
         <translation>&lt;hr/&gt;&lt;p&gt;Вы можете изучить функции программы без устройства, просто нажмите кнопку &lt;b&gt;Демо Режима&lt;/b&gt;.&lt;/p&gt;</translation>

spam-receiver, 2020-05-17:
<location filename="../src/main.cpp" line="81"/>
        <source>Demo mode without scope HW</source>
        <translation>Демонстрационный режим без аппаратного обеспечения</translation> to <translation>Демо режим без аппаратуры осциллографа</translation>

        <location filename="../src/selectdevice/selectsupporteddevice.ui" line="75"/>
        <source>Demo Mode</source>
        <translation>демонстрационного режима</translation> to <translation>Демо Режим</translation>

        <location filename="../src/selectdevice/deviceslistmodel.cpp" line="24"/>
        <source>Status</source>
        <translation>Status</translation> to <translation>Статус</translation>

        <location filename="../src/configdialog/DsoConfigFilePage.cpp" line="21"/>
        <source>Take a screenshot of the display area only</source>
        <translation>Сделайте снимок экрана только в области отображения</translation> to <translation>Делать снимок экрана только в области отображения</translation>

Ho-Ro, 2020-05-23:
Let's keep this issue open for future translation.

spam-receiver, 2020-05-23:
OK. Let it be. )))

Ho-Ro, 2020-05-26:
Hi, please check:

        <source>Set slowest possible timebase&lt;br/&gt;(&lt;b&gt;GUI may become very unresponsible!&lt;/b&gt;)</source>
        <translation>Установите самую медленную временную базу&lt;br/&gt;(&lt;b&gt;GUI может стать очень медленным!&lt;/b&gt;)</translation>

and

        <source>Minimal time between captured frames&lt;br/&gt;(Longer times reduce the CPU load)</source>
        <translation>Минимальное время между захваченными кадрами&lt;br/&gt;(Дольше времени снижает нагрузку на процессор)</translation>

spam-receiver, 2020-05-26:
```
<source>Set slowest possible timebase&lt;br/&gt;(&lt;b&gt;GUI may become very unresponsible!&lt;/b&gt;)</source>
<translation>Установите самую медленную временную базу&lt;br/&gt;(&lt;b&gt;GUI может стать очень медленным!&lt;/b&gt;)</translation>
```
 OK
```
<source>Minimal time between captured frames&lt;br/&gt;(Longer times reduce the CPU load)</source>
<translation>Минимальное время между захваченными кадрами&lt;br/&gt;(Дольше времени снижает нагрузку на процессор)</translation>
```
```
<translation>Минимальное время между захватами кадров&lt;br/&gt;(Чем больше времени, тем ниже нагрузка на процессор)</translation>
```

spam-receiver, 2020-06-12:
<location filename="../src/configdialog/DsoConfigScopePage.cpp" line="77"/>
        <source>Save settings on exit</source>
        <translation>Сохранять настройки при выходе</translation>

        <location filename="../src/configdialog/DsoConfigScopePage.cpp" line="81"/>
        <source>Save settings now</source>
        <translation>Сохранить текущие настройки</translation>

Ho-Ro, 2020-06-22:
@spam-receiver:  New text, please provide a good translation:

    <message>
        <location filename="../src/configdialog/DsoConfigScopePage.cpp" line="83"/>
        <source>Scope has hardware modification for AC coupling (restart needed to apply the change)</source>
        <translation type="unfinished"></translation>
    </message>

spam-receiver, 2020-06-22:
Осциллограф имеет аппаратную модифицикацию под переменный ток (для активации требуется перезапуск)

spam-receiver, 2020-06-22:
<message>
    <location filename="../src/configdialog/DsoConfigScopePage.cpp" line="83"/>
    <source>Scope has hardware modification for AC coupling (restart needed to apply the change)</source>
    <translation>Осциллограф имеет аппаратную модификацию под переменный ток (для активации требуется перезапуск)</translation>
</message>

spam-receiver, 2020-06-25:
[openhantek_ru.ts.gz](https://github.com/OpenHantek/OpenHantek6022/files/4828613/openhantek_ru.ts.gz)

spam-receiver, 2020-06-25:
[openhantek_ru.ts.gz](https://github.com/OpenHantek/OpenHantek6022/files/4830704/openhantek_ru.ts.gz)

Everything looks nice in Fedora 32.

spam-receiver, 2020-06-28:
[openhantek_ru.ts.gz](https://github.com/OpenHantek/OpenHantek6022/files/4842745/openhantek_ru.ts.gz)

Make work a hot key for checkbox "SP2" ("Спектр 2"). It was in conflict with menu "Help" ("Помощь").

spam-receiver, 2020-11-04:
[openhantek_ru.ts.zip](https://github.com/OpenHantek/OpenHantek6022/files/5484839/openhantek_ru.ts.zip)

Shortened the names in the math selection dropdown list ("К" instead of "Канал") to reduce the width of the "Voltage" block in Russian localization after https://github.com/OpenHantek/OpenHantek6022/commit/d845b2e0bf99dbb9dceab84d127a00410a8b6d64

closed 2021-03-12 by Ho-Ro