OpenHantek/OpenHantek6022 #93: Interface language switch

spam-receiver, 2020-05-19:
What about language selector in application?

spam-receiver, 2020-05-19:
I could`t find in translation file: "rms" (Vrms) and "pp" (Vpp).

Ho-Ro, 2020-05-20:
>> What about language selector in application?

What do you mean? The language is chosen according to the environment variable `LANG` (at least under Linux or Unix), also the windows OpenHantek.exe comes up with German text on a German Windows PC. Under Linux (my development system) you can start the program with `LANG=fr OpenHantek` to get the French version, with `LANG=C OpenHantek` you will get no translation at all.
When I started OpenHantek6022 as a fork of the old openhantek only `de` and a partly `pt` translation were available but not in use. I activated the automated translation and added a rough `fr` and `it`, `fr` was completed later by a French contributor, as well as the new `es` and `ru` (thx, :) ) versions.
As a digital storage scope is a complex technical object and the *lingua franca* of electrical engineering is English most foreign users are happy with english, therefore we never put high effort into complete *i18n*, because *trigger* or *channel* is understood by every engineer. Also the use of ASCII (or the latin character set) is quite standard, I think no one of us designed with Russian, Greek (or even more complicated Chinese) text in mind.

>> I could`t find in translation file: "rms" (Vrms) and "pp" (Vpp).

Ok, I forgot to *i18n* these strings. Nevertheless you can search in the code with e.g. `git grep '"rms"'` that will give:

    openhantek/src/dsowidget.cpp:                                                     "rms" );
    openhantek/src/exporting/legacyexportdrawer.cpp:                                  valueToString( result->data( unsigned( channel ) )->rms, UNIT_VOLTS, 3 ) + "rms",

The change `"rms"` -> `tr("rms")` will be visible in the next commit, please provide the correct ru *l10n*, I will use "Впп" so far but I don't know the correct "Vrms" abbreviation.

spam-receiver, 2020-05-20:
Export zoomed screen in double height - Экспорт увеличенного экрана с двойной высотой
rms (root mean square) - скз (среднеквадратичное значение)
CH1 - К1
CH2 - К2
MATH - Мат.
SP1 - Сп1
SP2 - Сп2
SPM - СпМ

I think that option of selection language in application`s menu is not such a bad thing. And you're right about the russian language in digital oscilloscopes. For me english version is most comfortable, because of 99% information about electronics I get in english. But there are some people who like everything in their native language. So I decided to help with my own russian language.

Ho-Ro, 2020-05-20:
>> I think that option of selection language in application`s menu is not such a bad thing.

What do you want to achieve? It's not so easy to use a menu setting because the l10n is enabled very early after program start long before the config data is available; but i can provide a command line option ("-i" or "--international") that disables the translation.

spam-receiver, 2020-05-20:
Command line option ("-i" or "--international") in Windows? It will be nice, thanks. I`ll check how russian translation looks in Fedora 32 Linux tomorrow.

spam-receiver, 2020-05-21:
<location filename="../src/docks/TriggerDock.cpp" line="26"/>
        <source>CH%1</source>
        <translation>К%1</translation>

        <location filename="../src/docks/TriggerDock.cpp" line="29"/>
        <source>CH%1 smooth</source>
        <translation>К%1 сглаженный</translation>

Ho-Ro, 2020-05-23:
Command line option "-i" or "--international", now in *unstable*, skips the translation.

closed 2020-05-23 by Ho-Ro

Anyeos, 2020-11-20:
> Export zoomed screen in double height - Экспорт увеличенного экрана с двойной высотой
> rms (root mean square) - скз (среднеквадратичное значение)
> CH1 - К1
> CH2 - К2
> MATH - Мат.
> SP1 - Сп1
> SP2 - Сп2
> SPM - СпМ
> 
> I think that option of selection language in application`s menu is not such a bad thing. And you're right about the russian language in digital oscilloscopes. For me english version is most comfortable, because of 99% information about electronics I get in english. But there are some people who like everything in their native language. So I decided to help with my own russian language. A few years ago I translated the original Hantek6022 program for Windows into russian.
> [Language_Rus_Eng.zip](https://github.com/OpenHantek/OpenHantek6022/files/4656401/Language_Rus_Eng.zip)

Same in Spanish, some words and paragraphs are too long so they can get truncated (and non understandable). And the window of the oscilloscope becomes too wide too. But that is unnecesary because I can have a more compact and little left panel if I use english and that is perfectly readable.