OpenHantek/OpenHantek6022 #94: making long time bases available through a command-line switch, without recompiling

bcrowell, 2020-05-21:
Hi,

Thanks for your quick response to my issue #92, "time bases longer than 100 ms," and for letting me know that this can be done by recompiling. Would it be possible to make this available through something like a command-line switch or something in the GUI? I'm using this scope with my community college physics electricity and magnetism students in the fall. We will be sending them lab kits to use during the covid-19 quarantines, and one item in the kit will be the Hantek oscilloscope. I can't expect them to recompile the software, so it would be really useful if they could have some way to turn this feature on without recompiling. Thanks!

-Ben

closed 2020-05-23 by Ho-Ro

bcrowell, 2020-05-26:
Thanks for making this change! I hadn't understood that you had actually done it until I was digging around in the code today and found this commit message: "allow to set the max timebase in scope settings menu from 100 ms/div to 1 s/div".

Ho-Ro, 2020-05-26:
Hi Ben,

I thought I wrote a comment about this change here, obviously I shut down the computer without pressing `Comment` :(
The big problem with long sample times is that the GUI freezes as long as one block of samples is recorded. 
I forked OpenHantek6022 in early 2019, after the maintainer stopped supporting the original OpenHantek that existed for about 10 years, so a lot of complex code has accumulated. A roll mode means deep restructuring in the data flow context, but I think it's worth a try - but only with low priority, let's see what it does ...

Martin

Ho-Ro, 2020-06-11:
> The big problem with long sample times is that the GUI freezes as long as one block of samples is recorded.

@bcrowell:
Hi Ben,
that behaviour annoyed me since the beginning - I refactored the USB capturing during the last week and now there is a *live mode* for very slow sampling. If you select trigger mode *None* the trace will be shown in near real time even at very slow sample rates (up to 5s/div -> 50 seconds on screen). The max time setting can be selected in the scope settings menu. 
Precompiled binaries of latest development status for all major OSs are always available in the [*unstable*](https://github.com/OpenHantek/OpenHantek6022/releases/tag/unstable) release.

Martin

bcrowell, 2020-06-11:
Yay, this is great, thanks! Playing with this, I found it a little confusing to understand what was going on, because when there is no signal (or just a DC signal), you don't get any visual indication of where the "beam" is currently located on the screen. On the Tektronix scopes I've used within recent memory, they just clear the whole trace when it gets to the right edge of the screen. 

It seems like you could also get rid of the need to activate long time bases through the Oscilloscope menu now. AFAICT that was only needed previously because the software used to become unresponsive on long time bases.

Ho-Ro, 2020-06-11:
Ben, this is WIP, stay tuned ...

Ho-Ro, 2020-06-15:
@bcrowell Please check the latest [unstable binaries](https://github.com/OpenHantek/OpenHantek6022/releases/tag/unstable).

commit 6d872a4d (2020-05-23): fix #94

allow to set the max timebase in scope settings menu from 100 ms/div to 1 s/div