OpenHantek/OpenHantek6022 #95: PDF Export - Waveforms not visible

cutephoton, 2020-05-25:
**Describe the bug**

PDF Output does not appear to have traces. (It does, but they are very very thin.) I am using a 4K monitor, but the main app seems to work well. Image Export is fine.

[image]
You can't see the waveforms...

Well... If you zoom in, the lines are visible, and very faint.
[image]

**To Reproduce**
It was consistent. Image png is rendered acceptably.

**Expected behavior**
Visible traces ;)

**Computer environment (please complete the following information):**
 - OpenHantek version: A very recent master (about a week old? as of May 25)
 - OS: Fedora 30
 - Video hardware: GeForce RTX 2060
 - OpenGL version: 4.6.0 NVIDIA 440.44

**Scope device (please complete the following information):**
 - Device 6022BE

Would like to express my appreciation for the work that has gone in to this application. I have a fancy scope for the real work, but this is a great tool for quick and dirty jobs.

Ho-Ro, 2020-05-25:
> Would like to express my appreciation for the work that has gone in to this application. 

Thx, I appreciate positive feedback - in contrary to an op amp that works better with negative feedback :)

> I have a fancy scope for the real work, but this is a great tool for quick and dirty jobs.

Me too, but my real 4-channel storage scope rests most of the time on the shelf, because it is much too bulky for my desk.

The image/pdf support has been an ugly hack in the original `Openhantek` for years, which I wanted to get rid of a year ago when I cloned my `OpenHantek6022`, because it had to duplicate all screen formatting (through OpenGL) with incompatible calls of the QPainter class, very hard to maintain consistency between screen and exported image.
Please use the new functions `Export/Screenshot` (makes a 1:1 picture of the complete program, either image or pdf),  `Export/Hardcopy` (takes a picture of the scope screen only with printer colors, either image or pdf) or  `Export/Print screen` to print directly. The new functions are available in the `unstable` release or the latest source code.
A zoomed screen can be exported in normal or double height, according to the setting in `Oscilloscope/Settings/File` menu. 
Printer colors (default) or screen colors for *Hardcopy* and *Print screen* can be chosen in `Oscilloscope/Settings/Colors` menu. 
You should delete (or rename) the old config file (~/.config/OpenHantek/OpenHantek6022) to use the new default printer colors.

P.S.: Windows users should use the system tool `regedit` and delete `Computer\HKEY_CURRENT_USER\Software\OpenHantek` to get the new default screen and printer colors.

Ho-Ro, 2020-06-01:
I assume you are satisfied with the solution, otherwise please reopen again.

closed 2020-06-01 by Ho-Ro